OpenPHDGuiding/phd2 #874: Guidestar wander

starhopper-de, 2020-08-12:
I am usind PHD more than a decade with great success, good work, many thanks!!!

My setup is a 12" f/4 newton on a Skywatcher EQ8 mount and a off-axis guider with a ASI 120 USB3 guiding cam. PHD Linux version and EQMOD. Guiding works perfect, no single frame fails.

I have attached a screenshoot, in which I marked the star position at the beginning of the session with a bookmark. After some time and dithers, the star wanders, with the result that I have to crop my image later. In this case there where 100 images (dithers). I am using random dither.

I suppose there maybe a bug. In my eyes the new position after dither should be calculated from the starting position, not from the recent position. Am I right, or is there another thing in my setup?

[image]

bwdev01, 2020-08-13:
Hi Thomas.  Dithers are always treated as relative offsets from the
*current* lock-position and it has always worked that way.  In fact, a
dither is created by simply changing the value of the lock-position and
letting the guide algorithms move the guide star to that location.  There is
no history kept of prior lock positions.  If you are doing large dithers and
are worried about the net drift of the lock position over long time periods,
you should use the spiral mode of dithering - that's really why that mode
was added.  In my experience, dithering nearly always results in some
cropping along the margins of the stacked frame because those areas
necessarily have a lower SNR than the rest of the frame.  
 
Cheers,
Bruce
 
  _____  

 

 
<https://user-images.githubusercontent.com/44210083/90057030-1a490700-dce0-1
1ea-8b5e-c331c38e25ef.jpg> PHD Star wandering
-
LTZBANCNFSM4P5FK26Q> .
<https://github.com/notifications/beacon/ADDHSV3JPUSRPZC7HV2OCULSALTZBA5CNFS
M4P5FK262YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4KDIICNA.gif>

starhopper-de, 2020-08-13:
Hello Bruce,
thank you very much for your detailed answer. My english is rather poor, but I understand, that this behavour is normal operation. I expected that new dither position were always calculated from the start position ( when I press the green button) of guiding or a special lock-position.
So, all star wandering resulting from refraction, bad mount and polar alignment accumulate during guiding pauses (autofocus, dither and so on). I will use spiral dither next time I had to do more than 50 frames per session. 

Cheers,
Thomas
www.starhopper.eu

closed 2020-08-16 by agalasso