OpenPHDGuiding/phd2 #876: PHD2 prints an error after Adjust Lock Position

nadalle, 2020-09-04:
Repro:

Tonight, I bookmarked my star, and then ran a calibration. After the calibration was done, I adjusted the lock position so that the star would move back to the bookmark, as it had drifted off significantly.

Actual behavior.

While PHD2 was emitting guide pulses to re-center the star, it printed an error message over and over like: "Unable to obtain sufficient motion in DEC" (or something similar).

It was moving in DEC perfectly fine, it just had a long way to move.

Expected behavior:

PHD2 should not print spurious errors, especially given that on the mailing list it was stated that clicking on the button to disable the error spam would also disable random features as well in some cases.

agalasso, 2020-09-04:
>  After the calibration was done, I adjusted the lock position so that the star would move back to the bookmark

@nadalle That sounds like you are looking for Sticky Lock Position behavior.  You can enable sticky lock position and PHD2 will keep the lock position at the starting position and won't move it after calibration completes. You can read about Sticky Lock Position in the PHD2 Help: https://openphdguiding.org/documentation/

> on the mailing list it was stated that clicking on the button to disable the error spam would also disable random features as well in some cases

could you please provide a link to the post containing the statement you are referring to so we can better understand what you mean and what you were told

agalasso, 2020-10-07:
assuming no response in over a month indicates this is not really an issue

closed 2020-10-07 by agalasso