OpenPHDGuiding/phd2 #878: Dark library rejected because of ASI bin/rounding issue

pludov, 2020-09-27:
I use ZWO ASI178MM for guiding with the indi driver. I am unable to use a dark library when binning 2. I spotted the following reason:

The row width for this camera is 3096. However that camera has a limitation were image width must be a multiple of 8. (can see that here: https://github.com/indilib/indi-3rdparty/blob/1ee8c5b96d7442b7235524ef023b14b4469d04e4/indi-asi/asi_ccd.cpp#L1181 )
So when taking a full frame in bin 2, the size ends up being 1544 (instead of the expected 1548...).

PHD2 works well with that limitation, appart for dark library: taking a dark library produces fits of width 1544, and then loading it is rejected with the following message:
> DarkLib check: failed geometry check - fits status = 0, cam dimensions = {1548,1040},  dark dimensions = {1544,1040}

I am unsure how this could be fixed, possibly : 
 * when setting the bin value, the driver report the effect available frame area - so we could use that instead of sensor size
 * record the actual sensor size in the dark library fit and compare that

agalasso, 2020-09-27:
@pludov thanks for the report, I'll find a way to fix that.

agalasso, 2020-10-05:
@pludov I did some work on this today, I think I have a fix. Are you able to test with this branch: #880 ?

pludov, 2020-10-06:
Thanks @agalasso 

I tested with indi driver and saw no apparent change :-( 

In this logs, PHD2 was started with a previous dark library, showing the problem, and I also acquired a new dark library, also generating the error.

[PHD2_DebugLog_2020-10-06_205054.txt](https://github.com/OpenPHDGuiding/phd2/files/5336081/PHD2_DebugLog_2020-10-06_205054.txt)

agalasso, 2020-10-06:
@pludov thanks for running the test.  I was expecting to see `FullSize=(0,0)` in the debug log right after the camera connects, but I see `FullSize=(1548,1040)` which makes me suspect your build may not have the changes from the branch.
Is there any chance that your build does not have the changes from the branch?  Did you do `git checkout indi_binned_frame_size` before you built?
I just pushed another change to the branch to update the phd2 version string so we can be sure we are working with the same code.
Could you pull and build and test again?  Please confirm the version is "2.6.9dev1-binframe1".
To get latest changes and build from the branch:
```
cd phd2
git fetch
git checkout indi_binned_frame_size
git pull -r
make -C tmp -j8
./tmp/phd2.bin
```
Also, could you test with verbose logging enabled:
[image]

pludov, 2020-10-06:
My bad : You are right, I didn't test with the version I built (2.6.6dev1 vs 2.6.9dev1). I'll retry tomorrow with the latest commit.

pludov, 2020-10-07:
It is working indeed ! I could not test on the sky, so can't judge the effectiveness of the dark...

agalasso, 2020-10-07:
ok, cool, I'll merge the fix to the main branch.

@pludov, if you have a chance would you mind running the same test with the ZWO native driver in PHD2 (make a new profile and select ZWO camera (not INDI))?  I made a corresponding change to the native driver which has the same problem.

Also, do you by any chance have a Windows computer? The problem exists for the ASCOM driver too. I have a change for the ASCOM case which I will push to the branch shortly.

pludov, 2020-10-07:
I tested with the ZWO native driver and it's working as well. (attached log file is with indi driver then asi driver after ~22:48 [PHD2_DebugLog_2020-10-07_205307.txt.gz](https://github.com/OpenPHDGuiding/phd2/files/5343786/PHD2_DebugLog_2020-10-07_205307.txt.gz) )

For windows/ascom I don't have that at home, so can't help.

agalasso, 2020-10-07:
ok, thanks for the help. I'll push the changes to the main branch. feel free to keep using your the build from this branch, there's nothing here that won't be in the main branch other than the version number change.

PR #880 (2020-10-07): ZWO / INDI / ASCOM cameras: do not assume binned frame size
Some ZWO ASI cameras break the assumption that the binned frame size is directly proportional to the un-binned frame.

Update the ZWO ASI native driver and the INDI camera client code to remove the assumption and use the actual binned camera frame to determine the binned frame size.

TODO: fix ASCOM cameras too

Fixes #878
closed 2020-10-07 by agalasso