OpenPHDGuiding/phd2 #884: PHD2's INDI panel works fine, but PHD2 connection button does not connect.

dzach, 2020-11-05:
Hi
I am building a DIY mount and have written an INDI driver which works fine with KStars, Stellarium, commandline utilities, and also with PHD2's INDI panel. But the gear button in the Connect Equipment dialog refuses to connect.

I suspect PHD2 expects some property that I' am not sending, but I have failed so far to find out which one it is. I've mostly copied the properties that seem relevant from INDIlib's Telescope Simulator, some of which are mentioned in other issues related to PHD2 connection problems, in the forum and in here. Looked also into the log files produced by PHD2, and made every change I thought relevant, but, no.

Here is a picture to make things merrier :-)
[image]

I attach a verbose log file from a PHD2 session.

[PHD2_DebugLog_2020-11-04_231810.txt](https://github.com/OpenPHDGuiding/phd2/files/5491097/PHD2_DebugLog_2020-11-04_231810.txt)

I would very much appreciate any hint about how to solve this problem.

It will be so nice for my DIY mount to afford the luxury of a PHD2 guide to the stars!
Thank you.

dzach, 2020-11-09:
So, I guess, my issue boils down to this:
What is the list of INDI properties (and their mandatory values, if any) that an INDIlib driver MUST present to clients, for a connection to PHD2 to be guaranteed?

bwdev01, 2020-11-09:
Have you studied the debug log?  What I see doesn't imply a problem with a missing property, it looks more like a basic comm failure.  Here's an excerpt:
23:19:31.472 00.000 139902604003072 INDI Mount: Received property: TIME_UTC
23:19:31.516 00.044 139902604003072 INDI Mount: Receiving message: 2020-11-04T21:19:31: [INFO] Azdeq is online. 
23:19:31.557 00.041 139902604003072 INDI Mount: Receiving Switch: CONNECTION = 1
23:19:31.557 00.000 139902604003072 INDI Mount: Receiving Number: FLIP_HA = 0  state = Ok
23:19:31.557 00.000 139902604003072 INDI Mount: Receiving Switch: PIER_SIDE = 0
23:19:31.557 00.000 139902604003072 INDI Mount: Receiving Switch: TELESCOPE_PIER_SIDE = 1
23:19:31.557 00.000 139902604003072 INDI Mount: Receiving Number: GUIDE_RATE = 0.5  state = Ok
23:19:31.690 00.133 139903230577344 INDI Mount: SideOfPier returns 0
23:20:01.497 29.807 139903230577344 INDI Mount bg connection failed canceled=0
23:20:01.497 00.000 139903230577344 Alert: Cannot connect to mount Azdeq: Connection timed-out
23:20:01.507 00.010 139902604003072 INDI Mount: serverDisconnected
23:20:01.507 00.000 139903230577344 INDI Mount: serverDisconnected

It looks to me like a connection was initially made, some property values were retrieved, then a connection failure occurred.  I don't think we're in a good position to help you debug your driver at this level.

dzach, 2020-11-09:
Thank you for the reply.
Yes, I have gone through it line by line. Actually, I have created the properties themselves, for my driver, by looking into the logs while connecting PHD2 with INDI's simulated telescope. I've also looked into INDIlib's verbose output line-by-line. Here is one from a session with PHD2. Here, I tried to connect immediately with the pre-configured driver, without previously invoking the INDI driver panel: 

[PHD2-INDI.log](https://github.com/OpenPHDGuiding/phd2/files/5512258/PHD2-INDI.log)

As it can be seen in INDI's log above, the driver responds to PHD2 by sending the property definitions, and then it responds to the CONNECTION property it receives by setting switch CONNECT On. Then it proceeds setting a few more properties, just as 'Telescope Simulator' does.

If invoked, PHD2's INDI dialog, shown in the picture above, connects immediately and controls the mount without a glitch.  So do KStars, Stellarium, CCDciel, INDIlib's command line tools, and just about everything I've thrown to it. So, if it is a communications problem, it must be something  pertaining to PHD2 connection procedure. Maybe some expected condition is not fulfilled and prevents it from connecting.

For that, I've tried to find my way into PHD2 source code, but don't seem to get very far, either.

bwdev01, 2020-11-09:
It should be pretty easy to see what PHD2 is doing - the Indi encapsulation
resides in scope_indi.cpp and you may want to start by looking at the
ConnectToDriver function.  I can see there are specific timeout conditions
there for various operations, perhaps that's the problem.  I'm not an INDI
guy, I don't know how to advise you.  If you post your question on the PHD2
forum instead of using GitHub, your issue will be seen by a wider audience.
 
Bruce
 
  _____  

connection button does not connect. (#884)
 

AU4PANCNFSM4TKWCRJQ> .
<https://github.com/notifications/beacon/ADDHSV6L7WOVEVNXR7BLCHTSPAU4PA5CNFS
M4TKWCRJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFMVBAGY
.gif>

dzach, 2020-11-09:
Thanks for looking into this Bruce.
I thought GitHub might be more appropriate, since this is a software development issue, and not a plain usage issue.
I'll try looking into the code again and report success here, in case someone else faces the same issue.
If that's not successful, I'll try the forum.

Dimitrios

bwdev01, 2020-11-09:
Looking at the code, PHD2 employs a 30-second timeout period for connecting
to the device.  The attempt to connect was logged at 23:19:31.46 which is
followed by a sequence of property returns of some kind.  The property
logging ended at 23:19:31.7 so that was done quickly.  But then there is
another 30 seconds with no log entries generated followed immediately by the
connection time-out message - that is the 30-second timeout being triggered.
AFAICT, the actual communication with the INDI server at this point is part
of connectServer(), which is part of baseclient.h in the INDI library.  So I
would guess connectServer() is waiting for a response that it's not getting
from your driver.
 
Bruce
 
  _____  

 
A7ABANCNFSM4TKWCRJQ> .
<https://github.com/notifications/beacon/ADDHSV7FTSKNEI4HRU7GEP3SPA7ABA5CNFS
M4TKWCRJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFMVNJ7Y
.gif>

dzach, 2020-11-09:
Which is exactly what baffles me, because INDI server's log shows that a response was indeed sent and passed along.

Let me re-iterate that there is no problem with the same exactly reply sent to PHD2's INDI panel (which connects immediately), or, as a matter of fact, with any other app I've used this driver with. I already have proper connection and continuous control of the mount with all these apps, including PHD2's INDI panel. This blocks my understanding of the problem, if it does indeed lie with INDI.

The way I read the specs, INDI (the XML protocol), signals a successful connection with a client requesting it, by setting the CONNECT switch of the CONNECTION property to 'On'. That's what my driver does. However, what a client does, when such a switch says its turned on, is a client's own business, not INDI's; in this case, its PHD2's business, and maybe of the resources it uses (some INDIlib version). I don't understand why PHD2 does not acknowledge reception of the <setSwitchVector> property, which appears in INDI server's log as being sent, while it does acknowledge connection when the same property is sent to the PHD2's INDI panel. Is the panel not using the same library / functions?

Thus, my question: what are PHD2's actual expectations from a driver, in terms of INDI properties, to concede that a connection was indeed made by the driver?

The only other issue related to communications that I can think of, is the character encoding used in the protocol's messages. I don't find anything in INDI spec, but the W3C XML spec states:
`
 All XML processors MUST accept the UTF-8 and UTF-16 encodings of Unicode`

I've tried UTF-8 (and ISO-8859-1, American ASCII, just in case), but they don't seem to change anything.

bwdev01, 2020-11-10:
Ok, maybe it will help to look at other examples that work correctly.
Attached is a doc that shows successful connection sequences from both EQMOD
and Avalon INDI drivers followed by the message sequence from your driver.
As I said, I don't know anything about the INDI protocol standard, but one
obvious difference is that you are returning both the property of CONNECTION
and later a "switch" message for CONNECTION.  The other drivers don't do
that, they only return the property.  
 
Beyond that, there may be threading differences between various applications
- the PHD2 connection logic is running on a background thread.  No idea if
that makes any difference to you, offhand I don't know why it would.  To
answer your other question, the code in the UI panel is separate from that
used in the GearDialog and no doubt behaves differently - the message
handling is different, there don't appear to be any protective timeouts,
etc.  
 
Bruce
 
 
 
  _____  

 
not acknowledge reception of the property, which appears in INDI server's
BS27ANCNFSM4TKWCRJQ> .
<https://github.com/notifications/beacon/ADDHSV6WOAAV4OMUGY5TTP3SPBS27A5CNFS
M4TKWCRJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFMWBVJQ
.gif>

dzach, 2020-11-10:
Thanks again for giving time to this issue, Bruce.
I'll go through the message sequence line-by-line, and copy it exactly, if necessary.
I understand that there is no need for any extra property to be included specifically for PHD2, other than those already included.

I can't see the attached doc. Was it attached?

bwdev01, 2020-11-10:
Let's try it this way - document attached here:
[Indi_Connection_Traffic.txt](https://github.com/OpenPHDGuiding/phd2/files/5514408/Indi_Connection_Traffic.txt)

bwdev01, 2020-11-10:
I don't know what's going on with GitHub, this is a really clumsy way to
communicate.  If you can send me a private e-mail from your normal address,
I'll send you the document.  E-mail me at bw underscore msg01 at
earthlink.net
 
Bruce
 
  _____  

 
CFYDANCNFSM4TKWCRJQ> .
<https://github.com/notifications/beacon/ADDHSV3MVG2OY2NZLD5MN23SPCFYDA5CNFS
M4TKWCRJ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFMWSJPI
.gif>

pchev, 2020-11-10:
Hi,
Sorry to look here only now, I need to review my Github notification.

The difference with other application is PHD2 check if the required guide command are available before to accept the connection. 
The check is done by CheckState() in scope_indi.cpp : https://github.com/OpenPHDGuiding/phd2/blob/master/scope_indi.cpp#L192

Normally as soon it receive both TELESCOPE_TIMED_GUIDE_NS and TELESCOPE_TIMED_GUIDE_WE it mark the connection ready and print a ready message with the available property we can see in the two example given by Bruce:
20:19:54.266 00.000 2978299952 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_NS
20:19:54.266 00.000 2978299952 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_WE
20:19:54.266 00.000 2978299952 INDI Telescope is ready MotionRate=0 moveNS=1 moveEW=1 guideNS=1 guideEW=1 coord=1
20:19:54.266 00.000 2978299952 INDI Mount: Received property: AUX1_FOCUSER_CONTROL

But this message is not present in your log:
23:19:31.471 00.000 139902604003072 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_NS
23:19:31.471 00.000 139902604003072 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_WE
23:19:31.471 00.000 139902604003072 INDI Mount: Received property: GUIDE_RATE
23:19:31.471 00.000 139902604003072 INDI Mount: Received property: PIER_SIDE

At this point it can be useful if you can run PHD2 step by step in debug to see why CheckState() do not work for you, or add more Debug.Write() in this function.

dzach, 2020-11-10:
I got it this time @bwdev01 
```
one obvious difference is that you are returning both the property of CONNECTION
and later a "switch" message for CONNECTION
```
Indeed, I added that last one in an effort to tell PHD2 the driver was connected.
There are two 'Connect' buttons at play here:

1. The 'Connect' button in PHD2's `Connect Equipment` dialog, which connects with the driver and turns into a 'Disconnect' button upon connection, and
2. The 'Connect' button in the actual INDI driver's page, which is paired with a separate 'Disconnect' button and connects/disconnects the mount.

What I send is the second 'Connect', which, rightfully as per the INDI protocol, is ignored by PHD2, since the driver itself is not yet connected. Removing that CONNECT switch does not make a difference.

dzach, 2020-11-10:
Thanks for replying @pchev 
Looking into `CheckState()` I see that the following flags/properties need to be true/exist:

```
IsConnected       : (CONNECTION ?)
m_ready           : ?
isAuxMount        : ?
MotionRate_prop   : (TELESCOPE_SLEW_RATE ?)
moveNS_prop       : TELESCOPE_MOTION_NS
moveWE_prop       : TELESCOPE_MOTION_WE
pulseGuideNS_prop : TELESCOPE_TIMED_GUIDE_NS
pulseGuideEW_prop : TELESCOPE_TIMED_GUIDE_WE
coord_prop        : EQUATORIAL_EOD_COORD
```
I assume the properties in parentheses correspond to the ones needed.
Will insert some Debug statements in each check and see where the checks fail.

dzach, 2020-11-10:
So, it looks like the failure is right at the beginning, when checking `IsConnected()`:
```
    if (!IsConnected()) {
	Debug.Write("INDI Telescope is not connected\n");
        return;
    }

15:03:58.617 10.400 140528551377600 gear_dialog: OnButtonConnectScope
15:03:58.617 00.000 140528551377600 Status Line: Connecting to Mount ...
15:03:58.619 00.002 140528551377600 Connecting to mount [INDI Mount [Azdeq]]
15:03:58.619 00.000 140528551377600 INDI Mount connecting to device [Azdeq]
15:03:58.626 00.007 140527934912256 INDI Mount: new device Azdeq
15:03:58.626 00.000 140527934912256 INDI Mount: Received property: CONNECTION
15:03:58.626 00.000 140527934912256 INDI Telescope is not connected
15:03:58.626 00.000 140527934912256 INDI Mount: Received property: ON_COORD_SET
15:03:58.626 00.000 140527934912256 INDI Telescope is not connected
15:03:58.627 00.001 140527934912256 INDI Mount: Received property: EQUATORIAL_EOD_COORD
15:03:58.627 00.000 140527934912256 INDI Telescope is not connected
15:03:58.627 00.000 140527934912256 INDI Mount: Received property: TARGET_EOD_COORD
15:03:58.627 00.000 140527934912256 INDI Telescope is not connected
15:03:58.627 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_SLEW_RATE
15:03:58.627 00.000 140527934912256 INDI Telescope is not connected
15:03:58.627 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_MOTION_NS
15:03:58.627 00.000 140527934912256 INDI Telescope is not connected
15:03:58.628 00.001 140527934912256 INDI Mount: Received property: TELESCOPE_MOTION_WE
15:03:58.628 00.000 140527934912256 INDI Telescope is not connected
15:03:58.628 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_ABORT_MOTION
15:03:58.628 00.000 140527934912256 INDI Telescope is not connected
15:03:58.628 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_TRACK_MODE
15:03:58.628 00.000 140527934912256 INDI Telescope is not connected
15:03:58.629 00.001 140527934912256 INDI Mount: Received property: TELESCOPE_TRACK_STATE
15:03:58.629 00.000 140527934912256 INDI Telescope is not connected
15:03:58.629 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_TRACK_RATE
15:03:58.629 00.000 140527934912256 INDI Telescope is not connected
15:03:58.629 00.000 140527934912256 INDI Mount: Received property: PIER_SIDE
15:03:58.629 00.000 140527934912256 INDI Telescope is not connected
15:03:58.629 00.000 140527934912256 INDI Mount: Received property: GUIDE_RATE
15:03:58.629 00.000 140527934912256 INDI Telescope is not connected
15:03:58.629 00.000 140527934912256 INDI Mount: Received property: DEBUG
15:03:58.629 00.000 140527934912256 INDI Telescope is not connected
15:03:58.630 00.001 140527934912256 INDI Mount: Received property: DEBUG_FILE
15:03:58.630 00.000 140527934912256 INDI Telescope is not connected
15:03:58.630 00.000 140527934912256 INDI Mount: Received property: FLIP_HA
15:03:58.630 00.000 140527934912256 INDI Telescope is not connected
15:03:58.630 00.000 140527934912256 INDI Mount: Received property: POLLING_PERIOD
15:03:58.630 00.000 140527934912256 INDI Telescope is not connected
15:03:58.630 00.000 140527934912256 INDI Mount: Received property: MOUNT_TYPE
15:03:58.630 00.000 140527934912256 INDI Telescope is not connected
15:03:58.630 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_PIER_SIDE
15:03:58.630 00.000 140527934912256 INDI Telescope is not connected
15:03:58.631 00.001 140527934912256 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_NS
15:03:58.631 00.000 140527934912256 INDI Telescope is not connected
15:03:58.631 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_TIMED_GUIDE_WE
15:03:58.631 00.000 140527934912256 INDI Telescope is not connected
15:03:58.631 00.000 140527934912256 INDI Mount: Received property: DRIVER_INFO
15:03:58.631 00.000 140527934912256 INDI Telescope is not connected
15:03:58.631 00.000 140527934912256 INDI Mount: Received property: TELESCOPE_INFO
15:03:58.631 00.000 140527934912256 INDI Telescope is not connected
15:03:58.631 00.000 140527934912256 INDI Mount: Received property: TIME_UTC
15:03:58.631 00.000 140527934912256 INDI Telescope is not connected
15:03:58.682 00.051 140527934912256 INDI Mount: Receiving message: 2020-11-10T13:03:58: [INFO] Azdeq is online. 
15:03:58.723 00.041 140527934912256 INDI Mount: Receiving Switch: CONNECTION = 1
15:03:58.723 00.000 140527934912256 INDI Mount: Receiving Number: FLIP_HA = 0  state = Ok
15:03:58.724 00.001 140527934912256 INDI Mount: Receiving Switch: PIER_SIDE = 0
15:03:58.724 00.000 140527934912256 INDI Mount: Receiving Switch: TELESCOPE_PIER_SIDE = 1
15:03:58.724 00.000 140527934912256 INDI Mount: Receiving Number: GUIDE_RATE = 0.5  state = Ok
15:03:58.819 00.095 140528551377600 INDI Mount: SideOfPier returns 0
```

dzach, 2020-11-10:
In scope_indi.ccp, ScopeIndi::ConnectToDriver() I see a comment:
```    
// wait for the device port property
```
but no call to such a property. I am not currently sending any serial/tcp port property.

In the same function, I see:
```
 while (!connection_prop && ...)
```
Assuming this is the loop where PHD2 is stuck waiting for a connection, it is strange that the CONNECTION property is not recognized, since the log shows that INDI Mount receives the CONNECTION property immediately after recognizing the device.
```
15:03:58.626 00.000 140527934912256 INDI Mount: Received property: CONNECTION
```
Looking further down in the code, I see:
```
void ScopeINDI::newSwitch(ISwitchVectorProperty *svp)
{
    // we go here every time a Switch state change
    if (INDIConfig::Verbose())
        Debug.Write(wxString::Format("INDI Mount: Receiving Switch: %s = %i\n", svp->name, svp->sp->s));

    if (strcmp(svp->name, "CONNECTION") == 0)
    {
        ISwitch *connectswitch = IUFindSwitch(svp, "CONNECT");

        if (connectswitch->s == ISS_ON)
        {
            Scope::Connect();
        }
```
This indicates that in order for the connection to be successful, not only there must be a CONNECTION property, but its value should also be **On** .

Setting the switch CONNECT to On inf the definition, seems to have solved the problem.
The driver connects immediately.

dzach, 2020-11-10:
And this is where my misunderstanding was: there are no two Connect buttons at play here, but one. Both Connect buttons, in the **Connect Equipment** dialog, and in the **INDI driver panel**, have the same functionality, i.e. they both set the CONNECTION property.

All other apps present an INDI driver panel, where the user can Connect/Disconnect the mount. When closing the INDI panel, the driver remains connected. PHD2 also presents the user with the same panel, but upon exit from the panel, the mount gets disconnected, even if the Connect driver button was left On. After disconnection, my driver resets the CONNECTION proprty switch CONNECT to Off, while PHD2 expects it On. Upon pressing the Connect button in the  **Connect Equipment** dialog my driver sent the reset CONNECT switch in the Off position, while PHD2 was waiting to receive it in the On position.

I should probably have guessed that behavior,  had I thought that the 'Press Here Dummy' principle applied to the Connect button too.

Thanks everybody for helping me solving this issue

closed 2020-11-10 by dzach

dzach, 2020-11-10:
I'd like to contribute the following page to the wiki, if that's OK with the maintainers:

### Writting an INDI driver for PHD2
When writting an INDI driver to work with PHD2, please consider the following:

The `Connect` button in the `Connect Equipment` dialog corresponds to the `Connect` switch of the CONNECTION property.
PHD2 waits for this switch to be turned `On`, before proceeding with the rest of the program.

The following properties MUST be present, for PHD2 to recognize and successfully connect to and operate with the driver:
```
CONNECTION, CONNECT: On, DISCONNECT: Off
TELESCOPE_SLEW_RATE
TELESCOPE_MOTION_NS
TELESCOPE_MOTION_WE
TELESCOPE_TIMED_GUIDE_NS
TELESCOPE_TIMED_GUIDE_WE
GUIDE_RATE
EQUATORIAL_EOD_COORD
```
Additionally:
```
PIER_SIDE
```
should be set, for PHD2 to be able to correctly set the side-of-pier when crossing the meridian, otherwise the pier side should be taken care of manually.

pchev, 2020-11-10:
I not think this is specific to PHD2, every INDI driver have a CONNECTION CONNECT and DISCONNECT property to connect the driver to the device:
https://www.indilib.org/develop/developer-manual/101-standard-properties.html#h2-general

All the driver I know use this property and behave the same. 
When the driver is started but not connected to the device it present a minimal set of generic properties, including CONNECTION.
After you set CONNECT=On the driver connect to the device and only now it can detect what model the device is and what functionality it support. Only at this moment it can define and send all the property definition to the clients, because it is now possible to use them.
Look how the telescope simulator work, the "Motion control" tab with the pulse guide properties appear only after CONNECT=On.
And when the device is disconnected by setting DISCONNECT=On the driver delete all this functional properties so the applications know they can no more use them.

If I understand the problem, your driver was sending all the properties, including TELESCOPE_TIMED_GUIDE_xx, even it is not connected to the device. 
This is probably not good, what do you do if an application use this properties to send a command when the telescope is not connected?

dzach, 2020-11-10:
What is specific to PHD2 is that the Connect button is separated from the INDI driver panel, following the mandra 'Push Here Dummy'. In other software the Connect button appears along with a Disconnect one, as it does in PHD2's INDI panel. That confused me a lot. It also explains why my driver behaved correctly with PHD2's INDI panel, but failed to do so with the single Connect button. That's why I propose to have a new wiki page explaining that.

Two reasons for sending the bulk of INDI properties initially:
1. The bulk of properties are sent as <defTypeVector> in reply to a <getProperties>  request. In the early stages of development, I just dumped an XML file with all properties set to initial values (hence the CONNECT = Off switch), and turned the mount On by using the INDI panel. That worked fine and I proceeded developing the rest of the mount functionality.
2. When changing to what is considered common behavior, i.e. sending a few generic properties initially and then populating the panel upon connection, I got a core dump on DISCONNECT, when I deleted the additional properties. I haven't yet resolved that problem, but in order to continue, I have reverted to populating the panel with all the available properties from the beginning, thus eliminating the need to delete properties on DISCONNECT.

I plan to deal with the core issue after I more or less finish with the development of this driver. What remains is the CCD part, send BLOBs and so on. Hopefully, that can be debugged more easily now.

pchev, 2020-11-10:
This is to make the INDI driver in PHD2 to work the same as the other internal or Ascom driver that have no panel.
The INDI panel in PHD2 is only here for the case the device need extra setup, for example to setup tcp/ip connection. But for most use there is no reason to show it.

In CCDciel I make the Connect button to work the same as in PHD2, it first connect to the server then it automatically connect the device. There is no real reason to show the full INDI panel only to connect the devices.

dzach, 2020-11-10:
Yes, now that I've cleared my misunderstanding and the driver is working, everything is fine. _As a user_, I like the Push Here Dummy idea and the single Connect button, and I wouldn't change it. I don't think I'll open the INDI panel again, once I finalize the property values for this mount. CCDciel worked with my driver from the very start, without a problem.

This was a developer issue, not a user one. When developing a driver without a definite list of mandatory INDI properties, the task of linearly exhausting all possible causes for a problem at hand, by changing property values and sequence and trying again, is huge. A few words about how the process flows when connecting and a comprehensive list of mandatory INDI properties would go a long way in clearing misunderstandings, especially for people like me with first time experience on similar equipment.

I am glad I'm now able to use this wonderful software with my DIY mount. Thank you all.

agalasso, 2020-11-11:
> I'd like to contribute the following page to the wiki, if that's OK with the maintainers:

@dzach certainly we'd welcome any updates to the Wiki.  The Wiki is open for edits, no permission required (but thank you for asking.)