OpenPHDGuiding/phd2 #887: Linker Problem with QHY

HaraldStauss, 2020-11-26:
I tried to compile PHD2 on a Rapberry Pi Zero W (Debian Linux). At the end of the compilation process, when it links phd2.bin, an error showed up saying something like "undefined reference to 'QHY ... init()'.
Any idea how to fix this?
Thanks,

Harald Stauss

pchev, 2020-11-27:
Can you try to edit the file thirdparty/thirdparty.cmake 
at line 1321 remove ^armv7(.*)|

HaraldStauss, 2020-11-27:
Thank you for your reply. 

uname -a gives:
Linux raspberrypi 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6l GNU/Linux

HaraldStauss, 2020-11-27:
Dear Patrick,Thank you so much for your fast reply.How about adding:^armv6(.*)|I am trying that right now.Compilation take a couple hours on the zero.I will let you know once it's done.Thank you for all your great work on phd 2.Harald

pchev, 2020-11-27:
The case for armv6 is taken at line 1317 and it must work for armv6l: 
    if (CMAKE_SYSTEM_PROCESSOR MATCHES "^armv6(.*)")
      set(zwoarch "armv6")
      set(qhyarch "armv6")
      set(toupcam_arch "armel")

Maybe this is another error because of the version of the armv6 qhy library?
Can you give the full error message from the linker.

HaraldStauss, 2020-11-27:
Here is the full error message from the linker:

[ 85%] Linking CXX executable phd2.bin
/usr/bin/ld: CMakeFiles/phd2.dir/cam_qhy.cpp.o: in function `QHYSDKInit()':
cam_qhy.cpp:(.text+0x300): undefined reference to `GetQHYCCDSDKVersion'
/usr/bin/ld: cam_qhy.cpp:(.text+0x7c8): undefined reference to `EnableQHYCCDLogFile'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/phd2.dir/build.make:2248: phd2.bin] Error 1
make[1]: *** [CMakeFiles/Makefile2:851: CMakeFiles/phd2.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

Thank you for looking into that.

Best regards,
Harald

pchev, 2020-11-28:
OK, the problem is the QHY SDK for armv6 is too old and not include this function, this make it not compatible with the current qhyccd.h.

From the GIT history the armv6 libqhyccd.a last update is from 2017 with the QHY SDK V7.4.16. Current version in PHD2 is V2019.11.15.0.

The only solution is to find a more recent version of libqhyccd.a for armv6. I not use a QHY camera myself and I don't know where to find this files. 
But it will probably work if you can find one from QHY and replace the file in cameras/qhyccdlibs/linux/armv6/

HaraldStauss, 2020-11-28:
Dear Patrick:

Thank you for looking into this.
It seems like the newer QHY SKDs, including the V2019.11.15.0 come only with a single libqhyccd.a for RPI3 and RPI4. Maybe they don't support armv6 any more.
I will try to replace the armv6 libqhyccd.a with the newest libqhyccd.a (that apparently is meant for the RPI3 or RPI4) and see if PHD2 compiles and runs with this new library (but I doubt it). I will report once compilation is done (can take hours).

Thanks,

Harald

HaraldStauss, 2020-11-28:
Brief Update:

I downloaded libqhyccd_armv6.bin
from https://github.com/indilib/indi-3rdparty/tree/master/libqhy
and renamed the file to libqhyccd.a and moved it in the cameras/qhyccdlibs/linux/armv6/ folder.
With this file, it compiled without error. 
But when I start ./phd2.bin I get an error saying: "Illegal instruction".

I am not sure what to do next.
Regards,
Harald

pchev, 2020-11-28:
I look at this file in the INDI code, but I am not sure it is really a armv6 version. 
The CMakeLists.txt in the same directory use libqhyccd_armv6.bin when the processor match "armv+", so armv6 and armv7.  
The Illegal instruction error you get is probably because this is really a armv7 file.

To be sure you can try to compile INDI and see if the qhyccd driver can load. 
If this work you can compile PHD2 OPENSOURCE_ONLY and use the INDI camera.
Otherwise I fear the only solution is to get a more recent RPi.

HaraldStauss, 2020-11-29:
OK. I compiled the qhyccd driver from the INDI 3rdparty github. The same error "Illegal instruction" occurred. So it is confirmed QHY does not provide an armv6 driver any more.
I submitted a ticket on the QHY website and asked for an armv6 driver. Let's see if they respond. 
I guess it is time to move to a RPi 4 (or back to MS-Windows).
Thank you for all your help.
Harald

pchev, 2020-11-29:
Please tell us about QHY response.
If armv6 is no more available we need to remove this old library and corresponding configuration.

HaraldStauss, 2020-11-29:
Yes, I will. My hope is still that they will provide an SDK for the armv6 or make the source code available so we can compile it on a Raspberry Zero W. I really like the small form factor of the Raspberry Zero W.

Once more thank you so much for all your work on PHD2. It is highly appreciated.

Harald

HaraldStauss, 2020-11-30:
Here is a brief update: 
I just received this reply from QHY:

Hi,
Does Raspberry PI Zero use different architecture with RPI3 and RPI4?Is it ARM32 or ARM 64?
Best Regards,
QinXiaoXu 

Apparently, they were not aware of the armv6 vs. armv7 issue. I explained it to them and asked if they could compile the library for the armv6. Let's wait and see. Maybe they provide an armv6 library soon.

Regards,

Harald

HaraldStauss, 2020-12-01:
Here is another update:

I received an e-mail from QinXiaoXu from QHY. 
They currently do not have a RPI Zero device but they may obtain one and then compile the SDK on that device for us. So there is hope that we will get the QHY SDK library for the RPI Zero soon..

QHY is very responsive and willing to help and to provide us with the driver. I really appreciate their support.

Regards,

Harald

HaraldStauss, 2020-12-08:
Hi Patrick:
So QinXiaoXu from QHY compiled the SDK using the -march-armv6 compiler flag and provided me with the library. 
So I copied this library into the source code of PHD2 and was able to compile it. The linker error did not show up any more.
But I still received the error message "Illegal Instruction" when I started phd2.bin.
But now I am thinking, maybe the old library was still installed somewhere on my system and phd2.bin may have used the old library instead of the new one.
I have to do more testing.

pchev, 2020-12-08:
Hi Harald,
If you cannot find another old library it can be useful to know what give this illegal instruction.
Run cmake with -DCMAKE_BUILD_TYPE=Debug and make again.
Then run phd2.bin with gdb and get a backtrace when it crash.

Unfortunately I cannot help more because I only have an armv7 RPi2 for testing.

HaraldStauss, 2020-12-08:
Dear Patrick:

 

Thank you for your comments.

My hope is that the new library that QHY provided me with yesterday will actually work on the armv6.

I think an old library was still floating around on my system, throwing things up.

I will try gdb tonight.

 

The problem is that the PI zero is somewhat slow and it takes several hours for the compiler to complete. This slows down the debugging.

 

I will report after I have completed some more testing.

 

Best regards and CLEAR SKYS.

 

Harald

d33psky, 2020-12-08:
Hi Harald,

Try this to find the library :

ldconfig -NXp | grep -i qhy

On my astroberry I have this as output :
	libqhyccd.so.20 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libqhyccd.so.20
	libqhyccd.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libqhyccd.so

-- Hans

HaraldStauss wrote on 20201208:

HaraldStauss, 2020-12-08:
Hallo Hans:

 

Ja, das werde ich gleich probieren wenn ich von der Arbeit nach Hause komme.

(Yes, I will try this right away when I come home from work).

 

Thank you,

 

Harald

HaraldStauss, 2020-12-09:
Hi:

I can now confirm that the armv6 library that QHY sent me works on the Raspberry Pi Zero/ZeroW.
After replacing an old library that was previously installed on my system, I was able to access my QHY 5-III 462C camera through the INDI driver in PHD2, CCDCiel and in KStars/EKOS.
However, I was not able to access it through the native driver from within PHD2. Also, I noticed that the ZWO native driver did not show up in the Device settings Window. Both the ZWO and QHY cameras can be accessed through the INDI driver but not through the native driver. I wonder if I have messed up the thirdparty/thirdparty.cmake file in a way that it does not recognize the armv6 processor and therefore, does not configure PHD2 for the native drivers?

Because the Raspberry PI Zero is a not very fast, my goal is to setup a minimalistic guiding system that avoids the overhead of the INDI driver. My thought was to connect the guide camera (QHY or ZWO) through the native driver to PHD2 and use the ST-4 guide port of the camera to control the mount (to avoid the INDI driver). But currently the native drivers do not show up in PHD2 on the Raspberry PI Zero. Currently, I can only access the QHY or ZWO cameras through the INDI driver. It is probably some sort of configuration (cmake?) issue.

But it is great that the new library from QHY works on the armv6 processor. I have asked QHY if I can share the library with others, or if they want to include the armv6 library with the official SDK release. Once I hear back from QHY, I may be able to share the library with you.

Best regards,

Harald

pchev, 2020-12-10:
It is possible the qhy header in phd2 need to be updated for this last version of the library.
Do QHY send you .h files with the new library?
Otherwise you can try to replace the qhyccd*.h from indi-3rdparty/libqhy/ to phd2/cameras/

HaraldStauss, 2020-12-11:
No, they just sent me the new library without the header files.
Switching the qhy library in /usr/lib/arm-linux-gnueabihf with the one provided by QHY allows using the qhy cameras with KStars/Ekos. So the new QHY driver definitely works.

However, I now found out that the qhy library is not the only library from the thirdparty package of PHD2 that is incompatible with armv6. The libtoupcam library is also not compatible with the armv6 processor. I don't think the SDK for the ToupTek  Cameras includes a library for the armv6 processor. 

I wonder, what changes need to be made to the thirdparty.cmake file to only include the ZWO and QHY drivers? Is there some sort of cmake directive that can be used or does it require editing the cmake file?
Thanks,

Harald

HaraldStauss, 2020-12-14:
I was finally able to compile PHD2 on the RPI Zero W (armv6). I had to disable the driver file for the toupcam (which is incompatible with armv6). But now it works very nicely and I can use the native QHY and ZWO drivers without INDI. On the weekend I used the RPI Zero W for some imaging. First, I polar aligned the scope using the Polar Drift Alignment in PHD2 and then I used Cartes du Ciel to move to M42. I then used CCDCiel to plate solve via ASTAP and move M42 in the FOV. Finally, I took some images using CCDCiel. When I just had the first couple of images, the clouds came in and I had to stop the experiment. But everything worked fine. Thank you so much for all the software that you provide. Best regards, Harald.

pchev, 2020-12-17:
Good you can make it work.

It is good if we can make the fix in the PHD2 source so it work for other user and also make more easy for you to get a new version.
I not see new SDK available in teh QHY download page, the last one is still 20.11.26. 

Can you make a PR with the change you do?
If not can you send me the modified files so I can make a branch for testing.

HaraldStauss, 2020-12-17:
Dear Patrick:

This is the reply that I got from QHY:

Hi,
    Thanks for your test result.
    Yes,we will release the version of armv6,but "-march=armv6" can't work on all platform,you know we need public all platform libraries,so we need modify CMakeLists.txt,this need some time to test,before that we can offer armv6 library for you specially.
Best Regards,
QinXiaoXu 

It would be great if the PHD2 sources could be modified so that it compiles on the armv6 processor.

QHY has sent me the library files for the armv6 and I used these to compile PHD2 on the PI Zero. But they seem to not want me to provide the library files to others at this time. Apparently, they are working on incorporating the armv6 library in their SDK. But it may take them some time to complete this. 

I can send you the changes that I made to compile PHD2 later today when I come home from work.
Regards,

Harald

HaraldStauss, 2020-12-18:
Dear Patrick:

I don't exactly remember what changes I made. 
But I modified the thirdparty.cmake file (attached). I had to remove the references to the toupcam library because it is not compatible with the armv6 processor. I just deleted it. Maybe one can use some cmake function to only remove it if an armv6 is detected (I don't know cmake). 
I also replaced the qhyccd library (in cameras/qhyccdlibs/linux/armv6) with the one that QHY had compiled for me.
Not sure if I made some more changes.
Thanks,

Harald

[PHD2_Changes.zip](https://github.com/OpenPHDGuiding/phd2/files/5713352/PHD2_Changes.zip)