OpenPHDGuiding/phd2 #892: Multistar guiding issue

ralphwaid1, 2020-12-07:


ralphwaid1, 2020-12-07:
[PHD2_DebugLog_2020-12-04_183412.txt](https://github.com/OpenPHDGuiding/phd2/files/5650522/PHD2_DebugLog_2020-12-04_183412.txt)

Mount EQ6 . Guider is listed ascom in log file. Same setup used for months with released phd2 software. First time I have seen a Ascom X frame error.

ralphwaid1, 2020-12-07:
See attached

agalasso, 2020-12-07:
@ralphwaid1 "Invalid X Frame" is an error message coming from the ASCOM driver. You'll need to ask the ASCOM driver provider (QHY) what that means and how to fix it.

agalasso, 2020-12-07:
I am going to close this as we have nothing actionable on the phd2 side

closed 2020-12-07 by agalasso

ralphwaid1, 2020-12-07:
If you say so but I wonder why I only got it when I enabled multi star guiding. I have been using this hardware for months with current phd2 software and QHY ascom drivers without a issue. The native driver for QHY is buggy and they recommend using ascom. I verified the native driver as buggy so I have been running ascom only. No biggie. I can run without multi star guiding. I see if QHY can shed light on this.

pchev, 2020-12-07:
Just before the "Invalid X Frame" error , at 20:18:22, you get an alert "After 16.0 sec the camera has not completed a 1.0 sec exposure".
This is generally an indication of issue at the USB level and you need to fix that.
For more information see:
https://openphdguiding.org/manual/?section=Trouble_shooting.htm#Problems-_Camera_Timeouts
Post in the forum for help if you cannot get it solved.