OpenPHDGuiding/phd2 #899: Multi-Star Guiding - feedback on documentation

deix, 2021-01-22:
After reading the "docs" for the latest development build, I have the following fedback (suggestions for making it more specific):

1. "...some of the secondary stars may be rejected because they are too dim or have drifted outside the field of view..." - does the  algorithm also reject stars when they became saturated ? 

2. "...The multi-star algorithm uses the secondary stars to refine the centroid position and reduce its volatilty, so there is essentially no way for it to degrade guiding performance...." - what happens, when the primary star get either too dimm / drifts from field of view or became saturated ? Does the algoritm picks the "best" secondary star and re-use it as the primary ?

3. "...12 stars in the field of view that have adequate SNR.  No more than 9 of these will be used at any one time..." - can You add the visualisation of "not active" stars that have been picked by auto-select:
- usual rectangle around the primary star
- green circles around the "active" secondary stars (up to 9 stars)
- red circles around the "not active" secondary stars (up to 12 stars)

 4. "...to replace secondary stars that are lost or rejected for some reason..." - You could list here the the criteria used in the algorithm, like e.g.:
- below SNR treshold
- too close to edge / drifted outside the field of view
- drifted inside frame (e.g. comet, geostationary satelite, etc...)
- saturated 
- etc...

bwdev01, 2021-01-22:
Hi Michael, thanks for the comments.  The manual generally focuses on
information that is needed to *use* a feature.  What you're talking about
here is mostly implementation-level detail and is subject to change at any
time. And none of it has any effect on your ability to use the feature.  For
this level of detail, the source code is the best reference, one of the
benefits of open-source software.   I've provided answers to your specific
questions based on the current state of the dev3 release.
 
  _____  

documentation (#899)
 

No.  Stars typically don't "become saturated" inasmuch as the stars that
were chosen by auto-find weren't saturated or near-saturated to begin with.
The primary star must always be available and it won't be automatically
replaced or substituted for if lost.  There's good reason for that - a
common reason for losing the primary star is for something mechanical to
cause a large shift in the pointing position of the scope.  In that case,
plowing ahead with guiding and imaging isn't a good idea.  A lost primary
star needs to be handled by the user or the imaging application, including
re-centering the scope and re-starting guiding.  When guiding is restarted,
the auto-find function will be invoked and the star list will be rebuilt.
Also, PHD2 won't dither the primary star out of the field of view so that
isn't a scenario to worry about.
I don't think I see the benefit of cluttering up the display any further
because none of this provides actionable information.  Once the novelty of
the feature wears off, you should just let it do its thing. :-)
Cheers,
Bruce
HN2NANCNFSM4WPADFWQ> .
<https://github.com/notifications/beacon/ADDHSV7I2Q4PU7CFZT7Z56DS3HN2NA5CNFS
M4WPADFW2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4LZZKGSQ.gif>

deix, 2021-01-23:
Thank You for the answers to your specific questions Bruce - let's close this post since there are no actions to be taken on it.
/ Michal

closed 2021-01-23 by deix