OpenPHDGuiding/phd2 #902: Cannot connect my QHY5L-II-m camera on arm64 architecture

readme9txt, 2021-02-02:
Hi,

I have a QHY5L-II-M, I tried to use phd2 on the Raspberry Pi 4, it works fine on Ubuntu Mate (32-bit). Later, I installed Ubuntu Desktop 20.10 (64-bit) on the Raspberry Pi, but I couldn't find "QHY Camera" in the camera selection of PHD2. Any idea how to fix this?

Thanks！

SteveBz, 2021-02-02:
Did you install phd2 like this (I typed it, so check for typos)?

sudo apt -y install libwxgtk3.0-dev

cd ~
git https://github.com/OpenPHDGuiding/phd2.git
mkdir -p ~/phd2/tmp  # Not sure what the -p does :)
cd ~/phd2/tmp
cmake ..
make
sudo make install

Then it should be just "QHY Camera" from the phd2 dropdown.

Mine works perfectly on my RPi3b+, but not Ubuntu, and my Kubuntu 20.04 desktop. Which is a branch of Ubuntu.

Good luck 

Steve.

readme9txt, 2021-02-02:
> Did you install phd2 like this (I typed it, so check for typos)? sudo apt -y install libwxgtk3.0-dev cd ~ git https://github.com/OpenPHDGuiding/phd2.git mkdir -p ~/phd2/tmp # Not sure what the -p [...]

Hi Steve:

Thank you for your reply.
Yes, I followed these commands to build phd2. I also tried to install phd2 from ppa:pch/phd2, but I couldn’t see the "QHY Camera" option in phd2, only "QHY5". 
I can't find armv8 in the phd2/cameras/qhyccdlibs/linux folder. Is it because QHY does not support armv8 architecture?

SteveBz, 2021-02-02:
Hi,

So it looks like that file is the QHY SDK.  Maybe it just hasn't been compiled for armv8 yet.  A little surprising.

These guys here are quite responsive:

https://www.qhyccd.com/bbs/#

They're on different times zones obviously, but if you go to them, I'm sure they'll recompile it for you, or tell you how to.  Maybe they just don't have an RPi4 :(

(Would it be too simplistic to imagine that you could just rename the armv7 directory to armv8?  I don't know how different the architectures are.)

Good luck.

Steve.

readme9txt, 2021-02-02:
> Hi, So it looks like that file is the QHY SDK. Maybe it just hasn't been compiled for armv8 yet. A little surprising. These guys here are quite responsive: https://www.qhyccd.com/bbs/# They're on [...]

I think I found a solution. I downloaded sdk_Arm64_21.02.01 from qhyccd.com, then copied sdk_Arm64_21.02.01/usr/local/lib/libqhyccd.a to phd2/cameras/qhyccdlibs/linux/armv8, and finally rebuilt phd2. 
Now it works fine :)

SteveBz, 2021-02-02:
Good job,  well done.

Maybe it can be uploaded to git too :)

Regards

Steve.