OpenPHDGuiding/phd2 #904: Dither with version 2.6.9Dev3

richardshopkins, 2021-02-13:
I was imaging last night and am stacking the images. I used APT to sequence the images and dithering was selected.
Despite random dithering being selected, looking at the individual offsets in deepskystacker, it seems that the dither is not entirely random but is systematically biased / drifts in both X and Y.
Individual frames are perfect, there is no drift and 5 minute exposures show pixel level guiding is working fantastically, the drift seems to occur in dither / between frames.
I have attached the guidelog. It shows the dithers, and the numbers do indeed "look" random.
I have also attached the deep sky stacker log which shows that the individual frames drift systematically but with some jumps. Over the imaging session, the overall drift was ~75-100 pixels in both axes.

Is this expected behaviour?
It would appear that for a very long duration sequence, there is a risk of the image drifting out of frame.

[PHD2_GuideLog_2021-02-12_183811.txt](https://github.com/OpenPHDGuiding/phd2/files/5975482/PHD2_GuideLog_2021-02-12_183811.txt)

[image]

bwdev01, 2021-02-13:
Hi Richard.  We have a very active forum for doing end-user support and we
would prefer that you use that.
 
https://groups.google.com/g/open-phd-guiding
 
What you're describing is an operational problem with your set-up, it isn't
an issue with the PHD2 code.  If you will post your question on the forum,
we can probably help you out.  You should also describe the mechanical setup
you're using for guiding because this sounds like it's probably differential
flexure.  I'm going to close this issue and wait to see your post on the
forum.
 
Regards,
Bruce 
 
  _____  

 

<https://github.com/OpenPHDGuiding/phd2/files/5975482/PHD2_GuideLog_2021-02-
12_183811.txt> 
 
<https://user-images.githubusercontent.com/79004335/107847941-b8c93880-6de7-
11eb-867c-a95503916830.png> image
-
ZJXPANCNFSM4XSAVKVQ> .
<https://github.com/notifications/beacon/ADDHSV5E2IAUPDRI4CGGHQ3S6ZJXPA5CNFS
M4XSAVKV2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4MBE4ZQQ.gif>

closed 2021-02-13 by bwdev01

richardshopkins, 2021-02-13:
OK Bruce,
Thanks for that. I will raise it there.
Richard