OpenPHDGuiding/phd2 #914: EventMonitoring: get half flux diameter

marcocipriani01, 2021-03-20:
Hello,
I'm creating an app to connect to PHD2 from the smartphone. I would like to take the phone outside with me and focus the guidescope reading the HFD value and looking at the live view, and maybe just check if the guiding is doing OK without going to the computer.
Getting the image from PHD2 was an easy task, but the EventMonitoring interface doesn't report the HFD value when looping exposures. I only get the HFD when guiding. Would it be possible to add this value to the `LoopingExposures` event if a star is selected?

If someone is interested, here's a screenshot of the app. I can release it to Google Play once is done.

[image]

Kind regards,
Marco Cipriani

agalasso, 2021-03-20:
hi @marcocipriani01 sure, we can add that, good idea.

Let us know when the app is ready and we'll put an announcement on openphdguiding .org -- and feel free to announce it in the PHD2 google group forum

marcocipriani01, 2021-03-20:
Thank you very much!

marcocipriani01, 2021-03-20:
To start, I've included the PHD2 control inside my app Telescope.Touch, already available on [Google Play](https://play.google.com/store/apps/details?id=io.github.marcocipriani01.telescopetouch). The app controls INDI telescopes, but if you just want to test the connection with PHD2, you can add a host in the connection tab and then go to the PHD2 module. No need for INDI.
At the moment, these functions are available (more to come):
- Live view with stretching
- Exposure selection
- Start guide/loop
- Graph

I will eventually make it a standalone app for those who use ASCOM, and then announce it!

marcocipriani01, 2021-03-20:
By the way, I've got an idea for PHD2, let me know what you think...
Instead of stretching the live view with a slider, I decided to implement a simple auto-stretch in the app. Here's how it works:

```
// 8-bit bitmap image
Bitmap bitmap = Bitmap.createBitmap(width, height);
// Temporary matrix
int[][] img = new int[width][height];

int min = Integer.MAX_VALUE, max = Integer.MIN_VALUE;
for (int w = 0; w < width; w++) {
    for (int h = 0; h < height; h++) {
        int val = /* read a pixel value from the PHD2 socket */;
        img[w][h] = val;
        if (val > max) max = val;
        if (min > val) min = val;
    }
}

// Determine scaling factors
double logMin = Math.log10(min), multiplier = 255.0 / (Math.log10(max) - logMin);
for (int w = 0; w < width; w++) {
    for (int h = 0; h < height; h++) {
        int stretched = (int) ((Math.log10(img[w][h]) - logMin) * multiplier);
        bitmap.setPixel(w, h, Color.rgb(stretched, stretched, stretched));
    }
}
// Show the bitmap to the user...
```

It's Java, but it should be pretty easy to understand. The resulting image is way better than PHD2's gamma-corrected preview, and stars are easier to identify.
Do you think this could be implemented in PHD2, too?

agalasso, 2021-03-20:
> I've got an idea for PHD2, let me know what you think...

@marcocipriani01  good idea and worth discussing, but please open a new feature request so we can keep this one on-topic

SteveBz, 2021-03-21:
Nice app, I just downloaded it. 

Steve. 

⁣Get BlueMail for Android ​

PR #915 (2021-03-28): event server: add star info to the LoopingExposures event
add the star info (mass, HFD, SNR) to the event server LoopingExposures event

Fixes #914
closed 2021-03-28 by agalasso