OpenPHDGuiding/phd2 #919: Improve status bar display for 'paused' state

bwdev01, 2021-04-17:
From user:
When PHD2 is set in pause, then the status bar is showing Pause for about 11 seconds but then the statusbar returns back to indicating Guiding. But it is still in pause. Would it not be better to show continuous something like "Guiding (paused)"  to avoid confusion about the status?

agalasso, 2021-04-17:
I'm not seeing the Pause status go away in that status bar. Can anyone else reproduce this?  We also have the PAUSED indicator in the bottom left of the image display (which also does not go away.)

[image]

han-k59, 2021-04-19:
There is a different statusbar message behavior for "pause of the output" or "full pause"

When a full pause is requested then the PHD2 status bar message change after 11 seconds from Pause back to Guiding. See image below.

For full pause the status bar should stay in Pause and for pause of the output I would suggest something like "Guider output paused". This to avoid any confusion.

Han

This is the situation after 11 seconds for full pause:
[image]

Form the documentation:
_When setting paused to true, an optional second parameter with value "full" can be provided to fully pause phd, including pausing looping exposures. Otherwise, exposures continue to loop, and only guide output is paused. Example: {"method":"set_paused","params":[true,"full"],"id":42}_

PHDcontrol program
12:11:28.481 00.000 5980 evsrv: cli 07CDDBF8 request: {"method":"set_paused","params":[true],"id":1}
12:11:28.481 00.000 5980 SetPaused type=1 isPaused=0 exposurePending=1
12:11:28.481 00.000 5980 Guider::SetPaused(1)

CCDCiel:
12:21:06.659 01.900 8436 evsrv: cli 04B55898 request: {"method":"set_paused","params":[true,"full"],"id":2002}
12:21:06.659 00.000 8436 SetPaused type=2 isPaused=0 exposurePending=1
12:21:06.659 00.000 8436 Guider::SetPaused(2)

commit f9528fc5 (2021-04-25): Clarify 'paused' status message. Fixes #919
closed 2021-04-25 by bwdev01