OpenPHDGuiding/phd2 #923: Add event server interfaces for features relating to star-selection

bwdev01, 2021-05-17:
1. Set/reset multi-star guiding
2. Get #stars in current star list (1 for single-star)
3. Set/reset sub-frames
4. Set/get MinHFD

agalasso, 2021-05-17:
@bwdev01 that sounds reasonable. Do you want to make a PR, or would you like me to do it? If you do make a PR, I'd be glad to review it.

bwdev01, 2021-05-17:
@agalasso - why don't you do it if you have time.  It's all your code and you will get it done 10x faster.  This may only be a partial list so you should consider additions.  For example, there is no way I see to set auto-exposure properties.  Regarding item 2, I'm just thinking of returning GuiderMultiStar::m_guideStars.size() which will work as well for single-star operation.  Obviously this will be a volatile number.  I'm trying to think through use-cases for smart automation apps to manage in star-poor regions.