OpenPHDGuiding/phd2 #932: Request to not force c++ standard on linux

mattiaverga, 2021-07-10:
Currently, phd2 forces c++ standard to be at max c++11:
https://github.com/OpenPHDGuiding/phd2/blob/771918f27b11ce5513b7a46def4ae70e5d6f48fc/cmake_modules/compiler_options.cmake#L61-L84

C++14 has been the [default standard](https://gcc.gnu.org/projects/cxx-status.html) since GCC 6 and with GCC 11 the standard is now c++17.

While maintaining the package in Fedora repositories, we prefer not to force a specific standard (I was required to do so at the time I submitted the package to Fedora repositories), so we [simply use](https://src.fedoraproject.org/rpms/phd2/blob/rawhide/f/phd2_2.9.10_std_cflags.patch) whatever default is preferred by GCC.
Apart from some warnings, I don't have any compilation error by doing so. Here are the build logs with [GCC 10](https://kojipkgs.fedoraproject.org//packages/phd2/2.6.10/1.fc33/data/logs/x86_64/build.log) (c++14) and [GCC 11](https://kojipkgs.fedoraproject.org//packages/phd2/2.6.10/1.fc35/data/logs/x86_64/build.log) (c++17).

Do you know of any shortcoming by doing that? Would you consider to drop that constraint?

agalasso, 2021-07-10:
@mattiaverga it looks like you have already identified what would need be changed. Do you want to send a PR?
The only concern I would have is to not break the Mac builds (since the else block starting on line 68 applies to both Linux and Mac) -- if you have a PR I can test it on Macs.

mattiaverga, 2021-07-11:
@agalasso I tried to make a PR which hopefully should not have impact on Mac builds.
From what I've seen from some online sources, on Mac it needs both `-std=c++11 -stdlib=libc++` while before the `-stdlib=libc++` overwrote the `-std=c++11` call... not sure about that.