OpenPHDGuiding/phd2 #935: Cannot connect to INDI server

rkaczorek, 2021-07-21:
Users are reporting this issue since dev5 version and it persists in 2.6.10
It has been reported that:

> PHD2 2.6.10 fails to connect to INDI. Oddly enough, if you use an existing PHD2 profile, it will still work after printing a "Mount Connect Failed" message... but I don't think you can create new PHD2 profiles.

I have just verified it and I can confirm that PHD2 in the latest stable version cannot connect to INDI server.

SteveBz, 2021-07-21:
Hi Guys, 

That's tragic. I'm on dev3, and I was about to upgrade. I can confirm that I can still connect to Indi. 

Kind regards, 

Steve 

⁣Get BlueMail for Android ​

rkaczorek, 2021-07-22:
I have just reviewed the source code and noticed that thirdparty directory contains libindi version 1.7.3 which is at most outdated (current version is 1.9.1). There's a lot of major changes introduced to libindi since 1.7.3 which was released June 2020.
Simple replace libindi and recompile does not fix the issue so the problem sits somewhere else.

mr-wacha, 2021-07-22:
FWIW - the log indicates that PHD2 is actually making succesful connection and can correctly enumerate devices supported by INDI, it's just that GUI doesn't respond / present user with choice when creating new profile.
Also, PHD2 would randomly drop connection to guide cam, and this symptom actually made me want to re-create the profiles (which turns out to be broken, as we know only now :-( ).
`11:51:08.668 05.440 3009577584 CameraFactory(INDI Camera)
11:51:08.668 00.000 3009577584 GetBoolean("/profile/2/camera/UseSubframes", 0) returns 0
11:51:08.668 00.000 3009577584 GetInt("/profile/2/camera/ReadDelay", 150) returns 150
11:51:08.669 00.001 3009577584 GetInt("/profile/2/camera/gain", 95) returns 95
11:51:08.669 00.000 3009577584 GetInt("/profile/2/camera/TimeoutMs", 15000) returns 15000
11:51:08.669 00.000 3009577584 GetInt("/profile/2/camera/SaturationADU", 0) returns 0
11:51:08.669 00.000 3009577584 GetBoolean("/profile/2/camera/SaturationByADU", 1) returns 1
11:51:08.670 00.001 3009577584 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.000000
11:51:08.670 00.000 3009577584 GetInt("/profile/2/camera/binning", 1) returns 1
11:51:08.670 00.000 3009577584 GetString("/profile/2/indi/INDIhost", "localhost") returns "localhost"
11:51:08.670 00.000 3009577584 GetLong("/profile/2/indi/INDIport", 7624) returns 7624
11:51:08.672 00.002 3009577584 GetString("/profile/2/indi/INDIcam", "INDI Camera") returns "INDI Camera"
11:51:08.673 00.001 3009577584 GetLong("/profile/2/indi/INDIcam_ccd", 0) returns 0
11:51:08.673 00.000 3009577584 GetBoolean("/profile/2/indi/INDIcam_forcevideo", 0) returns 0
11:51:08.673 00.000 3009577584 GetBoolean("/profile/2/indi/INDIcam_forceexposure", 0) returns 0
11:51:08.726 00.053 2800668720 INDIConfig: newDevice QHY CCD QHY5LII-M-6097e
11:51:08.739 00.013 2800668720 device QHY CCD QHY5LII-M-6097e interface(s) CCD|GUIDER|FILTER
11:51:08.739 00.000 2800668720 INDIConfig: newDevice Toupcam ATR3CMOS09000KPA
11:51:08.740 00.001 2800668720 device Toupcam ATR3CMOS09000KPA interface(s) CCD|GUIDER
11:51:08.740 00.000 2800668720 INDIConfig: newDevice Telescope Simulator
11:51:08.742 00.002 2800668720 device Telescope Simulator interface(s) TELESCOPE|GUIDER
11:51:08.743 00.001 2800668720 exclude device Telescope Simulator not a valid camera
11:51:18.422 09.679 3009577584 INDI Verbose Logging enabled
11:51:23.959 05.537 2800668720 INDIConfig: newDevice QHY CCD QHY5LII-M-6097e
11:51:23.977 00.018 2800668720 device QHY CCD QHY5LII-M-6097e interface(s) CCD|GUIDER|FILTER
11:51:23.977 00.000 2800668720 INDIConfig: newDevice Toupcam ATR3CMOS09000KPA
11:51:23.978 00.001 2800668720 device Toupcam ATR3CMOS09000KPA interface(s) CCD|GUIDER
11:51:23.979 00.001 2800668720 INDIConfig: newDevice Telescope Simulator
11:51:23.979 00.000 2800668720 device Telescope Simulator interface(s) TELESCOPE|GUIDER
11:51:23.979 00.000 2800668720 exclude device Telescope Simulator not a valid camera
`

bwdev01, 2021-07-22:
Sorry you’ve run into problems.  We will look into this as soon as possible – probably not until the weekend.  Until then, we don’t really need any more “me too” messages – it sounds like the problem can be easily reproduced.

 

Bruce

KrisAstro, 2021-07-23:
Hi,

you could use the cmake directive -DUSE_SYSTEM_LIBINDI=1 like this : 

cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Release -DOPENSOURCE_ONLY=1 -DUSE_SYSTEM_CFITSIO=1 -DUSE_SYSTEM_LIBUSB=1 -DUSE_SYSTEM_EIGEN3=1 -DUSE_SYSTEM_LIBINDI=1 ..

Make sure you have define in LDFLAGS the complet path to the libindi (ex : export LDFLAGS="-L/usr/local/lib").

I just tested, and that work fine (i only tested with indi simulator).

Kris

agalasso, 2021-07-25:
@rkaczorek I wanted to start looking into this but I cannot reproduce the problem with either an existing profile or a new profile -- both work fine and connect to the INDI server without any issues AFAICT.  I am testing on a Manjaro x86_64 system with KStars 3.5.3.

We'll need more information to diagnose the issue you were seeing.

1. system info: what linux distro, version, and cpu arch? What version of KStars? Any other INDI server info?
2. what are the exact steps that you did to reproduce the problem? What was the error message or other indication that show that it is not working.
3. please enable verbose logging in the indi camera or mount connection window, run the test scenario, then include a link to the uploaded PHD2 Debug Log (Help > Upload Log Files); you may have to select the option "Show logs with no guiding" in the log upload window.

mr-wacha, 2021-07-25:
Hopefully that'll do it for now until @rkaczorek will be able to add/amend.

1. Raspbian GNU/Linux 10 (buster) on RPI4. Astroberry 2.0.3. INDI 1.9.1 KStars/Ekos 3.5.4
'''astroberry@astroberry:~ $ sudo apt list --installed | grep kstar

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

kstars-bleeding-data/unknown,now 6:3.5.4 all [installed,automatic]
kstars-bleeding-dbg/unknown,now 6:3.5.4 armhf [installed,automatic]
kstars-bleeding/unknown,now 6:3.5.4 armhf [installed,automatic]
astroberry@astroberry:~ $ sudo apt list --installed | grep indi 

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

indi-aagcloudwatcher-ng/unknown,now 1.6-1 armhf [installed,automatic]
indi-aok/unknown,now 1.1-1 armhf [installed,automatic]
indi-apogee/unknown,now 1.9-9 armhf [installed,automatic]
indi-armadillo-platypus/unknown,now 1.0-9 armhf [installed,automatic]
indi-asi/unknown,now 1.9-5 armhf [installed,automatic]
indi-astroberry-amh/unknown,now 0.3-7 armhf [installed,automatic]
indi-astroberry-diy/unknown,now 2.8-4 armhf [installed,automatic]
indi-astroberry-piface/unknown,now 2.2-7 armhf [installed,automatic]
indi-astromechfoc/unknown,now 0.2-5 armhf [installed,automatic]
indi-atik/unknown,now 2.6-6 armhf [installed,automatic]
indi-avalon/unknown,now 1.12-1 armhf [installed,automatic]
indi-bin/unknown,now 1.9.1-1 armhf [installed,automatic]
indi-bresserexos2/unknown,now 1.0-1 armhf [installed,automatic]
indi-celestronaux/unknown,now 0.7-5 armhf [installed,automatic]
indi-dreamfocuser/unknown,now 2.1-9 armhf [installed,automatic]
indi-dsi/unknown,now 0.4-3 armhf [installed,automatic]
indi-duino/unknown,now 1.15-1 armhf [installed,automatic]
indi-eqmod/unknown,now 1.0-6 armhf [installed,automatic]
indi-ffmv/unknown,now 0.3-6 armhf [installed,automatic]
indi-fishcamp/unknown,now 1.1-9 armhf [installed,automatic]
indi-fli/unknown,now 1.5-7 armhf [installed,automatic]
indi-full/unknown,now 1.9.1 armhf [installed,automatic]
indi-gphoto/unknown,now 3.0-6 armhf [installed,automatic]
indi-gpsd/unknown,now 0.5-10 armhf [installed,automatic]
indi-gpsnmea/unknown,now 0.2-9 armhf [installed,automatic]
indi-maxdomeii/unknown,now 1.3-9 armhf [installed,automatic]
indi-mgen/unknown,now 0.1-9 armhf [installed,automatic]
indi-mi/unknown,now 1.8-7 armhf [installed,automatic]
indi-mqtt/unknown,now 1.0.8-1 all [installed]
indi-nexdome/unknown,now 1.5-5 armhf [installed,automatic]
indi-nightscape/unknown,now 1.0.6-6 armhf [installed,automatic]
indi-pentax/unknown,now 1.0-7 armhf [installed,automatic]
indi-qhy/unknown,now 2.7-7 armhf [installed,automatic]
indi-qsi/unknown,now 0.9-9 armhf [installed,automatic]
indi-sbig/unknown,now 2.1-9 armhf [installed,automatic]
indi-shelyak/unknown,now 1.0-9 armhf [installed,automatic]
indi-starbook-ten/unknown,now 0.1-1 armhf [installed,automatic]
indi-starbook/unknown,now 0.8-9 armhf [installed,automatic]
indi-sv305/unknown,now 1.2.3-3 armhf [installed,automatic]
indi-sx/unknown,now 1.15-8 armhf [installed,automatic]
indi-talon6/unknown,now 2.0-9 armhf [installed,automatic]
indi-toupbase/unknown,now 0.5-3 armhf [installed,automatic]
indiwebmanagerapp/unknown,now 6:1.5 armhf [installed,automatic]
libayatana-appindicator3-1/stable,now 0.5.3-4 armhf [installed,automatic]
libayatana-indicator3-7/stable,now 0.6.2-3 armhf [installed,automatic]
libindi-data/unknown,now 1.9.1-1 all [installed,automatic]
libindi-dev/unknown,now 1.9.1-1 armhf [installed]
libindi1/unknown,now 1.9.1-1 armhf [installed,automatic]
libindicator3-7/stable,now 0.5.0-4 armhf [installed,automatic]
'''
2. Either "New Profile Wizard" on PHD restart (with all profiles removed) or "Add New" without PHD restart - when selecting "INDI Guide Camera" or "INDI Mount" while creating profile in the dialog "INDI Camera Selection" connection to INDI servers stays "Disconnected", clicking "Connect" yields no change and (obviously) the "Driver" drop-down stays empty. No error messages, other then "Disconnected" status in this dialog and upon dialog close typical mesage to say " PHD wasn't able to connect to the camera...etc"

pchev, 2021-07-25:
Hi,
I start to look at this strange issue.

Personally I have no problem with INDI 1.9.1 and PHD2 2.6.10 when running on Ubuntu x86-64 or i386.

I just update a RPi2 that run Raspbian Buster and add the Astroberry repository to install INDI 1.9.1 and PHD2 2.6.10.
I confirm I get the issue you mention:
- After a click to the profile setup button it show the INDI server as disconnected and cannot select the device. The log show it connect and get the device list.
- Using an existing profile it can connect after showing the error message.

Then I compile PHD2 from source using the instruction at https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux
- copy/past the apt-get block to install the Debian/Raspbian dependencies.
- create tmp dir
- run cmake with: cmake -DOPENSOURCE_ONLY=1 ..
- make

Running the compiled ./phd2.bin work normally without issue...
This is unfortunate because the next step I plan was to add some debug print to understand why it not continue after getting the devices list.

Can anyone confirm it work after compiling the program this way.

@rkaczorek  , one of the difference I remark between the two phd2.bin is the Astroberry version is compiled for GTK2 but my version is for GTK3. This can be controlled by installing libwxgtk3.0-gtk3-dev. This was changed because new Ubuntu versions no more support wxWidgets with GTK2. There is no change in the PHD2 code for that but it is possible we get a GTK2 bug.
Can you look at this or for any other difference in the Astroberry build process.

mr-wacha, 2021-07-26:
@pchev just confirmed that build with **-DOPENSOURCE_ONLY=1** connects to INDI as expected.
Unfortunately good weather spell just ended and it prevents me from testing actual guiding performance but I assume we are not concerned here.

One more thing to note that compiled version exhibits strange graphics artefacts (and I apologise if this is know, i.e. due to GTK library version difference)?
- the "+" buttons next to numerical fields are almost completely obscured
- "Star Profile" window background flickers
- in the "History" window (guiding graph), the RMS numerical data background flickers too (window background is grey-is, but data points have black background)

Do we want separate Issue for tracking of above or we understand it is effect of GTK library version difference?

pchev, 2021-07-26:
Thank you for testing and for the confirmation.

Please note that option -DOPENSOURCE_ONLY=1 I mention has nothing to do with the solution, it is here only to simplify the debug of the program by allowing to start ./phd2.bin directly without messing with proprietary library installation path because we not need this libraries to test the INDI connection.

The appearance problem are probably because of the GTK3 theme currently in use. 
I not know enough with Raspbian to give any guidance on how to adjust that, but on my Kubuntu system it work perfectly using the GTK standard "Adwaita" or "Adwaita-dark".

rkaczorek, 2021-07-26:
This is really strange.
I have upgraded to libwxgtk3.0-gtk3-dev and then compiled it manually with the following cmake parameters:
`cmake -j4 -DCMAKE_BUILD_TYPE=Release -DOPENSOURCE_ONLY=1 -DUSE_SYSTEM_CFITSIO=1 -DUSE_SYSTEM_LIBUSB=1 -DUSE_SYSTEM_EIGEN3=1 -DUSE_SYSTEM_LIBINDI=1 ..`
... no success
I have also compiled and packaged it using cdbs... no success.
Whatever I do I end up with GUI not responding to connecting to INDI. No access to INDI devices either.
Debug log attached.
[PHD2_DebugLog_2021-07-26_140934.txt](https://github.com/OpenPHDGuiding/phd2/files/6879253/PHD2_DebugLog_2021-07-26_140934.txt)

rkaczorek, 2021-07-27:
I would add that version 2.5.9dev4 works without problem. I have reviewed diff between the 2.5.10 and 2.5.9dev4 and see no changes that could introduce anything in the code except major translation updates. Also, just to confirm I'm using the latest dependencies in my build environment as indicated here https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux

rkaczorek, 2021-07-27:
I have nailed it! It only works if you use -DUSE_SYSTEM_LIBINDI=0 which means using INDI library (1.7.3) provided by PHD2 sources. Using -DUSE_SYSTEM_LIBINDI=1 i.e. INDI library installed in a system results in a bug in handling INDI devices.
Final conclusion is that PHD2 needs to be updated to use the latest INDI libraries.

pchev, 2021-07-28:
Thank you Radek, I get same conclusion after more testing on my RPi2. I have wrongly interpreted your third message here making me thing it also not work with 1.7.3 so I not first searched in this direction.

There is no problem to link PHD2 with the 1.7.3 INDI BaseClient code, the XML message content is still exactly the same when the server is 1.9.1 so it work perfectly. The quick solution is to use -DUSE_SYSTEM_LIBINDI=0 or not specify it.

I will try to reproduce the problem on my main computer to search for a solution in better condition because it take more than one hour to compile PHD2 on my RPi2.
From my first testing the problem is BaseClient no more call ServerConnected() after the connection is established.

rkaczorek, 2021-07-28:
Thanks for your support Patrick. Actually the third message is a result of wrong assumption on my side. I thought the problem comes from old libs whereas it's the other way around. Only after further investigation I realized that I used system libs each time at compilation and this causes the issue.

pchev, 2021-07-28:
Find it! 

This is because of a bug in INDI 1.9.1 that was fixed by Jasem three weeks ago:
https://github.com/indilib/indi/commit/3092e823d1f23b7d7f359ae296acfd22dfa54a73

I propose we keep 1.7.3 in PHD2 for now.
Then wait all the refactoring work on INDI is in a more stable condition before to get an update of the included client code.

pchev, 2021-07-28:
Just to confirm it work fine with INDI master.

agalasso, 2021-07-28:
@pchev thanks for tracking this down!

rkaczorek, 2021-07-28:
This is great news. Thanks everyone for tracking this down!
I have just reviewed INDI commits history and indeed the bug was fixed just after starting INDI 1.9.2 development cycle. So it exists in 1.9.1. This means that working version PHD2 2.5.9dev4 is just coincidence. This explains why I was not able to find any changes introduced to PHD2 that would break connection to INDI. I would suggest that we keep this issue open until stable INDI 1.9.2 is released so we can replace 1.7.3 libs with the latest ones.

pchev, 2021-07-29:
Yes it is good to keep this open as a reminder to update to 1.9.2 when available. I see there was a number of fix to the client code since one year so it is good to take them.

You can also keep wxWidgets-GTK2 as long it is available with Raspbian as my first intuition it was a form message issue is wrong.