OpenPHDGuiding/phd2 #938: Linux build: Advanced Settings "Camera" and "Guiding" blank

SebNik1, 2021-08-16:
Dear OpenPHD development team,

first thank you very much for your work in this project! 

I am using phd2 2.6.10 on an rpi4 8gb running ubuntu 20.04.2. I connect remotely via indi server. Everything works so far but I wanted to try multi-star guiding and I cannot enable that because the advanced settings "guiding" menu is all blank. I tried both installing the final phd2 package and building it from source (Aug 16th 2021).

[image]
[image]

Both pictures were taken with camera and mount fully connected. As far as I remember the menu was never shown since first setting up the rpi in september 2020. 

I looked in the documentation and searched the web, as far as I found out this souldnt be blank. If I am missing something I apologize. If needed I will try providing further information on your request. 

Thank you in advance!
Sebastian

bwdev01, 2021-08-16:
Hi Sebastian.  I don’t know anything about the Unix builds but I suspect it’s because of the color theme you’re using on the system.  Try changing that, perhaps setting it back to something basic.

 

Bruce

 

 <https://user-images.githubusercontent.com/36997718/129587562-9520ffed-5b7b-4c6d-bd0e-188923e2bffd.png> 

 <https://user-images.githubusercontent.com/36997718/129587579-3b06043e-a48b-4acb-9b66-4c84feeffe4f.png>

SebNik1, 2021-08-16:
Hi Bruce,

that was exactly the problem! It is very confusing because the „Global“ and „Algorithm“ tabs were displayed regularly and without any disturbance… so this is solved then.

Thank you very much!
Sebastian

closed 2021-08-22 by SebNik1