OpenPHDGuiding/phd2 #939: Properties not opening on PHD2

Ferwol, 2021-08-18:
Hi again.
First of all, thanks for all the work behind this proyect.
I am close to connect "arduinost4" to PHD2. In Ascom diagnostics everything seems fine (screenshot 1), Cartes du Ciel connects to st4 and I can set different COM port in properties.
But, in PHD2 I click on properties button and nothing happens. Trying to connect will send an error (screenshot 2) and if I try to set properties via Device Hub the error is different (screenshot 3).
Tried unistalling and installing Ascom, phd2 and driver several times. Checked registry with CCleaner. Also installed 1.1 driver version. Don´t know what else could I do. I am stuck.
Thank you again!!

[image]
[image]
[image]

bwdev01, 2021-08-19:
This doesn't look like a PHD2 problem, the missing file error message is being generated from the ASCOM Utilities module.  You should use the ASCOM device hub or POTH as a testing mechanism - if it doesn't work, PHD2 won't be able to connect either.  Also, please post problems and questions like this on our Google groups forum, not here in the repository.

Bruce

closed 2021-08-19 by bwdev01

Ferwol, 2021-08-19:
My intention was to post this on Arduino st4 GitHub Page. Sorry for the mistake!