OpenPHDGuiding/phd2 #950: needed a delay between the mount displacement and the

olivdeso, 2021-11-05:
On heavy setups, having a large inertia (like EQ8 for instance), is it common to see the mount still moving during the exposure.

conditions : guide speed x0.5, exposure 3s, mount loaded at its 3/4 max payload

-> it would be very usefull to add a tunable delay (few hundreeds of ms) between the end of the mount displacement and begening of the exposure. Default 0ms.

(unfortunatelly, the existing tunable delay at the end of the exposure is not usefull in that case)

bwdev01, 2021-11-05:
Please submit your PHD2 logs and identify a time when you think this occurred.  If the mount is still moving after the conclusion of the guide-pulse, that's a bug in the mount driver.  It's up to the driver to accurately signal that completion.  I think it's likely you've mis-diagnosed the problem.  If you're convinced you want to force a delay after the completion of the last guide command and before the next guide exposure, you can use the time-lapse setting on the Camera tab of the Advanced Settings dialog.  In future, it's best to post your questions/concerns on our PHD2 forum rather than on GitHub.

Bruce

closed 2021-11-05 by bwdev01

olivdeso, 2021-11-05:
Hi Bruce,

I think you miss understood my point : the ascom driver can't know the inertia of the mount nor its actual position with mounts that don't have precise absolute encoders mounted on axis and and controled throught an open loop motor control like steppers. Typicaly the EQ8. (it does not occurs with the EQ6 unless it is realy overloaded)
It depends on the inertial moment which is unknown to the driver. So The driver can't know when the mount precicely stops mooving. In fact the mount never stops moving exactly at the end of the guide pulse.

The workaround is to use a much slower guide speed. Thus the mount stops faster after the end of guide pulse.

-> I will try to get some examples with differents mounts. I noticed that mainly with the EQ8. I use 2 of them in Chile, one equiped with a Newton 350/1800 which has a quite large moment of inertia.

-> I will try the trick you said using the  time-lapse setting on the Camera tab

That said I also noticed the issue you described with some Celestron mounts : the mount still moves a little bit after "it said" it has stoped, but it is a different issue from my point here. It is often due to a problem with the DC servo encoders and play in gears.
This could occurs on other DC servo controled mount like Losmandy or AP or Vixen, but I never noticed that on AP (Mach1,900,1100). I had some issues with the early Losmandy Gemini 2, but solved since a long time.
But the Vixen SXD was never really stabilized.

clear skies

Olivier

bwdev01, 2021-11-05:
> displacement and the beginning of the exposure (Issue #950)

Ok, let's see if we can figure this out.  I probably don't quite follow your
reasoning here but that's ok - I'm not an engineer and I'm always willing to
learn something new.  But instead of debating your hypothesis, I think it
would be better to go back to a data set that illustrates your point.  In
your original message, you were asking for a delay mechanism between the
time of the last guide pulse and the start of the next guide camera exposure
(which the timelapse control will do).  That's really what's confusing me.
It sounds like you're saying that the guide camera exposures are showing
trailed stars because the mount is still moving with the drive motor off.
Have you captured guide camera images that I can see that show those trailed
stars?  Short of that, I can't really picture what the guide logs could show
that would lead you to your conclusion.  We've certainly seen over-shoots
and evidence of stiction on the Dec axis, but those are mechanical issues
that can't be dealt with in PHD2.
 
Anyway, I think it will work best if you can upload both your guide and
debug log files to our support forum here:
https://groups.google.com/g/open-phd-guiding
 
PHD2 has a feature for uploading log files, its under Help/Upload Logs.
Ttying to work through this via GitHub is fairly inconvenient for me so I'd
prefer to handle it on the support forum.
 
Thanks,
Bruce

  _____  

O5KNANCNFSM5HM6CWRQ> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/ADDHSV7YSDXGLATKRVQ7V7DUKO5KNA5CNFS
M5HM6CWR2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHFKCIRQ
.gif>