OpenPHDGuiding/phd2 #964: Feature request - Simultaneous RA and Dec corrections

MasterPIC, 2022-01-04:
Hi,
what about adding simultaneous RA and Dec corrections? This feature can improve guiding.
An option would be required to enable it as some controllers don't support it.

Clear Skies!
Armando

bwdev01, 2022-01-19:
It's not worth the complications IMO.  The amount of overlap you achieve is likely to be on the order of 100ms or less which isn't going to have any material effect on guiding.

MasterPIC, 2022-01-24:
It's a missing feature... There is no reason for waiting for RA correction end before sending Dec one.
I see only the need for waiting the max RA/Dec correction duration before starting the new guiding exposure.
For hardware able to guide simultaneously on RA and Dec, current guiding implementation is a pointless complication IMO.

d33psky, 2022-01-24:
I totally agree from a principal standpoint.
And I understand Brett's point on the probably minimal effect.
But I do not have the bandwidth to implement at this time either.