OpenPHDGuiding/phd2 #988: PHD2 Terminology  RMS error versus standard deviation

han-k59, 2022-05-25:
Probably an old topic but RMS error is not a fully adequate term for what PHD2 is reporting. What I see during software testing is that PHD2 reports the standard deviation:

**Standard deviation= square root( mean((errors - mean(errors))^2))**

 while

**The RMSE= square root( mean((errors)^2))  and the error is the deviation from a predicted or set value.**

I don’t think there is a predicted value or setpoint except the mean of the measured values. PHD2 clearly removes any DC component of the trend. So displaying SD or σ rather then RMS error would be more appropriate.

I tried to post it at the PHD2 forum but all my messages are deleted automatically. :(

Han

See also:
https://stats.stackexchange.com/questions/264906/rmse-vs-standard-deviation-in-population

bwdev01, 2022-05-28:
I think I’m ok with the way this is done and have a different view on the matter.  Essentially, there is a “predicted value” and that value is zero.  In that case, RMS and sigma produce the same results.  We’re not keeping track of the absolute positions of the stars on the sensor, we’re interested in the deltas between those positions and their associated reference locations.  The guide algorithms, in general, are not purely reactive, nearly all of them use past history as a factor in forecasting what guide pulse will restore the delta-values to zero.  This is analogous to doing a forecast and then measuring the differences between the projected and actual values – a typical use-case for computing standard deviation.

In any case, we wouldn’t want to change the UI and the documentation and then deal with hundreds of end-user questions about why it had been changed after all this time – especially when there is no implication at all to the actual guiding results which is where we keep our focus.  And as I said, I think it’s correct as-is.

closed 2022-05-28 by bwdev01

han-k59, 2022-05-28:
_>>Essentially, there is a “predicted value” and that value is zero._
The control algorithm needs an offset or delta to correct proportional (aggression). With a drifting DEC position this is clearly visible in simulation with PHD2 because DEC stays above zero or below zero. With a drifting DEC the offset can't be zero because then the drift would not be compensated.  So with the control setting you can influence the DEC position but it is a minute amount. This mean offset or DC component is ignored by the RMS error calculation because it calculated first the mean value of DEC position first before calculating the deltas.

_>>We’re not keeping track of the absolute positions of the stars_
If you don't keep track of the absolute star position in the image , then you can't guide. You need absolute positions to calculate the deltas.

Maintaining the status quo after so many years is understandable

han-k59, 2022-05-28:
Here an example screenshot of PHD2. The RMS error in DEC trend is clearly about 2.5" but PHD2 is reporting 0.38" only. So the standard deviation of the DEC trend.

[image]