OpenPHDGuiding/phd2 #999: phd2 failing to start

phomer60, 2022-07-16:
After building latest version from repository on Raspberry Pi OS 64bit Buster, the programs fails to start.

Not sure if there are any logs I can check.

Paul

agalasso, 2022-07-16:
@phomer60 what do you see if you run PHD2 from a terminal window?

phomer60, 2022-07-16:
/usr/bin/phd2.bin: error while loading shared libraries: libSVBCameraSDK.so: cannot open shared object file: No such file or directory

phomer60, 2022-07-17:
Dependencies checked as per instructions on [https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux](url) and everything is installed.

phomer60, 2022-08-06:
Since INDI now appears to run under Bullseye 64bit I am no longer going to worry about this.

jruizluque, 2022-10-03:
Take a look at this PR [https://github.com/OpenPHDGuiding/phd2/pull/1011](https://github.com/OpenPHDGuiding/phd2/pull/1011)

agalasso, 2023-12-26:
I believe this was resolved so closing it now.

closed 2023-12-26 by agalasso