OpenPHDGuiding/phd2 #1000: Since latest update: Calibration step (ms) gets set to random numbers

RononDex, 2022-08-11:
I have updated PHD2 (by compiling it from source) and since then, it appears that it is completly unable to calibrate the guiding.
This is because, the Calibration step gets a very random high or low number (like 1ms, or 8500ms).

I think the reason for this is, that for some reason since the update, it is unable to fetch some data from INDI / EKOS, the Calibration Calculator shows empty fields:
[image]

This is while the INDI mount is connected in PHD2 and it provides Location / pointing data.

Whenever I set the calibration steps, calibration distance and calibration declination manually and press ok, it gets reset the moment EKOS triggers a new calibration, so manual settings are useless.

pchev, 2022-08-11:
I just rebuild INDI and PHD2 from the last source but cannot reproduce this issue.

Can you indicate which source and commit you build, what option do you use for cmake, what INDI version you use.

But this look like an issue with communication with the INDI server. Can you upload the debug log using this procedure : https://openphdguiding.org/getting-help/  and provide the link here.

Can you you also try to start an INDI server with only the ccd and telescope simulator, create a new profile for them in PHD2, and try if you get the same issue.

RononDex, 2022-08-11:
My PHD2 was compiled on commit 2ae2cfa9, and I use following script to compile and install it:
```
InstallPHD2() {
    cd ~/packages/phd2
    mkdir -p compiled
    cd compiled
    cmake ..
    make
    sudo ln -sf /home/cobra/packages/phd2/compiled/phd2.bin /usr/bin/phd2
}
```

INDI Library has version 1.9.8

When I startup PHD2 it also shows this message at the top:
[image]

If I go to "upload log files" it does not show any log files from today

RononDex, 2022-08-11:
Update: Found the checkbox for logfiles without guiding:
https://openphdguiding.org/logs/dl/PHD2_logs_id33.zip

pchev, 2022-08-11:
Your are connecting to the camera directly using the ASI SDK. 
Are you sure this camera is not also connected to INDI by Ekos?

If it is you must select "INDI Camera" in PHD2 then select the ASI the same way you select the INDI mount.

RononDex, 2022-08-11:
Yes I am sure that it is not connected to INDI too, I have not added it to the connection profile of EKOS.

The logs show something about an error getting pierside of the mount, could that be an issue?

RononDex, 2022-08-11:
Just did a test with the guiding camera connected through INDI. The camera works, but still the same issues as described in the first post

RononDex, 2022-08-11:
What is even more wird: In the "camera" tab in PHD2 it shows the correct pixel size, but inside "Guiding" tab it still shows the wrong one:
[image]
[image]

pchev, 2022-08-11:
Yes I see this change between 3.75 and 0.1.

It look like sometime the profile return one or the other value:
12:39:41.099 00.030 3058946080 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 3.750000
12:48:14.558 00.030 3058946080 GetDouble("/profile/2/camera/pixelsize", 0.000000) returns 0.100000

I can reproduce that partially by editing the pixel size in the calculator when the camera is not connected, I can set 0.1 here. Then if I close PHD2 a pixelsize of 0.1 is saved to the profile. 
If I restart, connect the camera, open the Advanced setting, I see the correct pixel size in the Camera tab and the calibration is changed for a too high value. 
But if I open the calculator the pixelsize, declination,.. are all right and if I validate it store the new calibration step with the right value.

Maybe your profile file is corrupted and this prevent to save the new values?

Can you try to rename the file ~/.PHDGuidingV2 to something else when phd2 is stopped, then create a new profile.

RononDex, 2022-08-11:
I just tried deleting the file.

The first run wizard reappeared. After configuring everything the same issue still persists

bwdev01, 2022-08-11:
The camera pixel size is obtained from the driver once the camera is connected.  If the value is wrong or changes, you should be looking either at the driver or the connection to the camera.  PHD2 isn’t prepared to deal with a camera driver that lies about the pixel size.

 

Bruce

pchev, 2022-08-12:
One more think to check is the version of libasi installed by phd2 and indi. This is a frequent problem when using the native driver.
I remark now you not install phd2.sh but directly phd2.bin, this miss the libraries normally installed in /usr/lib/phd2, maybe try "make install" instead.

After I have no more idea. Maybe try to install the same on another computer using the packages for a comparison.

tstibor, 2022-08-12:
Probably the most cleanest approach is to build a PHD2 package. PHD2 enables via the build system to build e.g. a DEB package:

`mkdir pkg && cd pkg && cmake .. && make package`

The result is e.g. the following package:
```
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: phd2
CPack: - Install project: phd2 []
CPack: Create package
CPack: - package: /home/tstibor/dev/astronomy/phd2-test/pkg/phd2_2.6.11.20220812073648.0apollo_amd64.deb generated.
```

The Debian package building approach also works perfectly:
`[tstibor@apollo ~/dev/astronomy/phd2]$ dpkg-buildpackage -us -uc -b -rfakeroot`

The result is:

```
-rw-r--r--  1 tstibor tstibor    17571 Aug 12 09:36 phd2_2.6.11_amd64.buildinfo
-rw-r--r--  1 tstibor tstibor     1979 Aug 12 09:36 phd2_2.6.11_amd64.changes
-rw-r--r--  1 tstibor tstibor 14959856 Aug 12 09:36 phd2_2.6.11_amd64.deb
-rw-r--r--  1 tstibor tstibor 32482888 Aug 12 09:36 phd2-dbgsym_2.6.11_amd64.deb
```

RononDex, 2022-08-12:
I am pretty sure it is not related to the driver, since the "Camera" tab in PHD2 shows the correct Pixel size. This holds true for both, direct connection to the ASI camera and through INDI.

I will try the different way of installing it and see if it changes anything

RononDex, 2022-09-24:
I am still having this issue, I can not get phd2 to work anymore, even trying all of the above suggested steps

RononDex, 2022-09-25:
I can also confirm that this happens on multiple / different pc's, with my ASI ZWO 224mc

RononDex, 2023-06-19:
After switching to astroberry (from Arch Linux ARM) it is now working. I probably had misconfigured something in the OS