OpenPHDGuiding/phd2 #1001: Support for new ZWO cameras (SDK 1.2.4 released June 2022)

CSPhobos, 2022-08-14:
I have a new ZWO ASI 662MC camera and it is not working with the current PHD2 software, probably due to old SDK drivers as this camera was just released. Their new SDK was released about two months ago.

The ZWO SDK with the latest drivers is available here: https://download.astronomy-imaging-camera.com/download/asi-camera-sdk-linux-mac/

bwdev01, 2022-08-14:
Ok, we’ll look into this.  In the meantime you should be able to use the ASCOM camera driver or you could even try using the updated SDK file.  The latter would basically be a “smoke test” – it might work or it might not depending on whether there are breaking changes in the new SDK.

 

Bruce

agalasso, 2022-08-28:
fixed by #1005

closed 2022-08-28 by agalasso