OpenPHDGuiding/phd2 #1020: Feature request: adjust lock position or sticky lock position in the API

gnthibault, 2022-12-22:
From the [documentation](https://openphdguiding.org/man-dev/Tools.htm#:~:text=the%20bookmark%20location.-,Lock%20Positions,off%20by%20a%20few%20pixels.) I see that it should be possible to force to change the position of a star (when selected) so that it is moved up to a specific coordinates on the sensor.

Unfortunately, from the [API documentation](https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#available-methods), and some tests, I didn't managed to find how to force a previsously selected star (with find_star) to go onto a specific position on the sensor.
This feature is particularly useful in slit-spectroscopy workflow.
I would summarize the feature request as follow:
* add a sticky_lock_position that would make the current lock position sticky, even in case of calibration
* add a adjust_lock_position, with an x,y parameters so that one can change

bwdev01, 2022-12-22:
Have you done this:

 

 

Bruce

gnthibault, 2022-12-22:
Sorry I don't understand your message.

bwdev01, 2022-12-22:
My message showed a screen-shot of the documentation for the set_lock_position method.  So my question is: 1) Did you know about it, 2) Did you try using it, and 3) Do you have some reason to think it doesn’t work.

d33psky, 2022-12-23:
(the screenshot did not show up for me either, not in mail nor on github)

bwdev01, 2022-12-23:
Ok try this.

<html><body>

set_lock_position | X: float, Y: float, EXACT: boolean (optional, default = true) | integer (0)
-- | -- | --

</body>
</html>

gnthibault, 2022-12-23:
yes sorry I forgot to add this in the initial problem description, I did quite some tests with set_position_lock .
The behaviour is the following: set_position_lock (almost) immediately returns with the required LockPositionSet event, it does not change the position of the previously slected star whatsoever, no mount movement is initiated at all.

Upon launching guide afterwards, phd2 selects another star (which generates another LockPositionSet), so I really think the feature is not available from the API.

bwdev01, 2022-12-23:
Ok, sorry, this looks like a bug wherein the freshly set lock-point is almost immediately over-written.  I can reproduce it easily enough so I should be able to deal with it soon.  The intended behavior is that the change in lock-point will behave like a dither, which is essentially all a dither is.  Doing a ‘start-guide’ will select a new star set unless there is already one in use.

 

Bruce

gnthibault, 2022-12-23:
> Ok, sorry, this looks like a bug wherein the freshly set lock-point is almost immediately over-written.  I can reproduce it easily enough so I should be able to deal with it soon.  The intended [...]

Exactly ! I would expect ideally the set_lock_position to perform a "dither" in the pixel space that would bring the current selected star onto the position passed in parameter in the set_lock_position.
It would be absolutely awesome if you could bring that feature up, I am ready to test this. Thanks a lot for your help

bwdev01, 2022-12-23:
Ok, I looked into this and I don’t think there’s a problem after all.  The “secret sauce” is that you must set the “Exact” parameter to true!  If you leave it set to false (the default), it behaves like “lock on the star nearest to this position”.  That’s useful if you have an app that’s trying to choose a guide star for some reason rather than a fixed sensor position.  But if there’s no star at that exact position, the next “star-find” of the closest one will reset the lock point.  With “Exact” set to true, it means “set the lock point here and move the guide star to this position” – just like a dither.  Here’s an excerpt from the log file showing the desired behavior:

 

14:59:23.289 00.002 7684 evsrv: cli 0D73BE08 request: {"method":"set_lock_position","params":[358,185,true],"id":1}

14:59:23.289 00.000 7684 setting lock position to (358.00, 185.00)

14:59:23.291 00.002 7684 Mount: notify guiding dithered (3.8, 3.6)

 

Give that a shot, I’ll bet it works…

 

Bruce

gnthibault, 2022-12-27:
Ok thank you very much @bwdev01  for actively providing support.
I think I was missing a few elements of understanding with regard to how PHD2 is working. Apparently while simply being in looping state, set_lock_position was immediately returning, not doing much, even if star was selected before (through the use of find_star)

However, when in state guiding, the set_lock_position indeed triggers a mount move towards the new target, which is great.
I haven't however, really understand if there was a specific event associated with the lock position being reached, I would have expected a SettleDone event, that would have been great.

However, I found a simple workaround, by simply monitoring guidestep events, and waiting for RADistanceRaw" and DECDistanceRaw to be as close as expected.
Just a side note, I notive there was a small issue in the documentation:
https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#guidestep
**Dec**DistanceRaw should be **DEC**DistanceRaw
**Dec**DistanceGuide should be **DEC**DistanceGuide

agalasso, 2022-12-28:
@gnthibault thanks for letting us know about the error in the documentation.  I made the correction, but the wiki is actually editable by anyone with a github account so feel free to make any additional corrections yourself!

gnthibault, 2022-12-28:
Sure thing, I will continue develop my python software, but there is a high chance that I split the Phd2 client into its own python module at some point.

I will ask my friend who is expert in spectroscopy to better understand the workflow, and potentially consider help developing new features if needed (I will most probably ask for support to develop here)
Will close the FR, thank you veruly much again for the support

closed 2022-12-28 by gnthibault