OpenPHDGuiding/phd2 #1032: Cannot compile PHD2 using latest INDI 2.0

paolostivanin, 2023-01-31:
Hello,
PHD2 up until 2.6.11dev4 can't be compiled when using system INDI 2.0 (released a few hours ago).

You can find the full build log here: https://build.opensuse.org/build/Application:Astrophotography/openSUSE_Tumbleweed/x86_64/phd2/_log

Thank you!

pchev, 2023-01-31:
Hi,
I cannot access your log without an account.
Can you confirm this two points:
1. you compile with "-DUSE_SYSTEM_LIBINDI=1"
2. the error is about override, like:
   scope_indi.cpp:118:10: error: ‘void ScopeINDI::newLight(ILightVectorProperty*)’ marked ‘override’, but does not override

paolostivanin, 2023-01-31:
Hello,
1. yes, I use the following options:

```
    -DUSE_SYSTEM_CFITSIO=ON \
    -DUSE_SYSTEM_LIBUSB=ON \
    -DUSE_SYSTEM_GTEST=ON \
    -DUSE_SYSTEM_EIGEN3=ON \
    -DUSE_SYSTEM_LIBINDI=ON
```

2. sorry, I thought it was freely accessible. The errors are about override. For example:

```
[  104s] /home/abuild/rpmbuild/BUILD/phd2-2.6.11dev4/cam_indi.cpp:169:10: error: 'void CameraINDI::newBLOB(IBLOB*)' marked 'override', but does not override
[  104s]   169 |     void newBLOB(IBLOB *bp) override;
[  104s]       |          ^~~~~~~
[  104s] /home/abuild/rpmbuild/BUILD/phd2-2.6.11dev4/cam_indi.cpp:170:10: error: 'void CameraINDI::newSwitch(ISwitchVectorProperty*)' marked 'override', but does not override
[  104s]   170 |     void newSwitch(ISwitchVectorProperty *svp) override;
[  104s]       |          ^~~~~~~~~
[  104s] /home/abuild/rpmbuild/BUILD/phd2-2.6.11dev4/cam_indi.cpp:171:10: error: 'void CameraINDI::newNumber(INumberVectorProperty*)' marked 'override', but does not override
[  104s]   171 |     void newNumber(INumberVectorProperty *nvp) override;
[  104s]       |          ^~~~~~~~~
[  104s] /home/abuild/rpmbuild/BUILD/phd2-2.6.11dev4/cam_indi.cpp:172:10: error: 'void CameraINDI::newMessage(INDI::BaseDevice*, int)' marked 'override', but does not override
[  104s]   172 |     void newMessage(INDI::BaseDevice *dp, int messageID) override;
[  104s]       |          ^~~~~~~~~~
[  104s] /home/abuild/rpmbuild/BUILD/phd2-2.6.11dev4/cam_indi.cpp:173:10: error: 'void CameraINDI::newText(ITextVectorProperty*)' marked 'override', but does not override
[  104s]   173 |     void newText(ITextVectorProperty *tvp) override;
[  104s]       |          ^~~~~~~
```

pchev, 2023-01-31:
It look like INDI 2.0 introduce major change in the client library that break compatibility.
I not think I am competent to fix this kind of issue, maybe @agalasso can take a look? 

In the mean time you can compile PHD2 with the included client library ( -DUSE_SYSTEM_LIBINDI=0 ). This work perfectly, even to connect to a 2.0 server, because the protocol itself do not change.

paolostivanin, 2023-02-01:
Hi! Thanks, I'll look into this solution while compatibility is being worked on.
Not sure I will be able to "officially" make that change though (packaging policy is strict).

tstibor, 2023-02-01:
The INDI library was refactored and redesigned, however kept compatability for older version via
 variadic definition [#if INDI_VERSION_MAJOR < 2](https://github.com/indilib/indi/blob/431869ffe400053fea79f227f80b175c593b3db8/libs/indidevice/indibase.h#L138)
I guess one could copy the latest FindINDI.cmake from the INDI project, where the version information is obtained during
the cmake run and then set properly the variadic definition: INDI_VERSION_MAJOR in the CXXFLAGS in the PHD2 project.

paolostivanin, 2023-02-01:
Ah nice, good to know! Later today I'll try and, if it works, I'll open a PR here!
Thanks a lot!!

pchev, 2023-02-01:
Thank you Thomas.
Do you know some documentation or discussion that describe this change more globally?
Using a deprecated flag can be a temporary solution but we also need to make a permanent fix for version 2.

tstibor, 2023-02-01:
Yes you are right, it is a temporary solution to get PHD2 compile against INDI Client 2.0.0.
Actually the biggest INDI 2.0.0 change is on the server and driver side. From the client side and PHD2 point of
view it is INDI commit: [Built-in shared pointers for INDI::BaseDevice (https://github.com/indilib/indi/pull/1764)](https://github.com/indilib/indi/commit/4a15b8ea4b5b4616aa85dc02bdc385cc93819985)

where e.g. `virtual void newDevice(INDI::BaseDevice *dp) // deprecated` => `virtual void newDevice(INDI::BaseDevice` baseDevice);`. So checking [INDI_VERSION_MAJOR < 2](https://github.com/indilib/indi/blob/431869ffe400053fea79f227f80b175c593b3db8/libs/indidevice/indibase.h#L138) and updating to the new pointerless function signature could a proper path.

paolostivanin, 2023-02-01:
I played around with both IVM<2 and fixing the code, but without any luck so far.

tstibor, 2023-02-01:
@paolostivanin Sorry I forgot actually, if latest INDI master that is version 2.0.0 is compiled and installed then

```
[tstibor@herkules /usr/include/libindi] grep -r MAJOR indi*
indiapi.h:#define INDI_VERSION_MAJOR   2
indibase.h:#if INDI_VERSION_MAJOR < 2
```

So the compatability for INDI_VERSION_MAJOR < 2 cannot be accessed. In other words, PHD2 when compiled against INDI 2.0.0 must follow the new signture `newDevice(INDI::BaseDevice baseDevice)`....

paolostivanin, 2023-02-01:
I'm trying to update the code, let's see how it goes!

paolostivanin, 2023-02-01:
I've looked a bit into this. The required changes don't seem too difficult to implement, but there are quite some of them.
I can try to work on this, but it will take me a bit to get up to speed

pawel-soja, 2023-02-07:
See https://github.com/OpenPHDGuiding/phd2/issues/1036

agalasso, 2023-02-19:
fixed by fixed by #1033 + #1040

closed 2023-02-19 by agalasso