OpenPHDGuiding/phd2 #1036: Mount connection times out with INDI 2.0

jpaana, 2023-02-07:
Tried using system INDI (current 2.0 GIT version) after #1033 was applied. Everything compiles but mount connection times out when starting PHD2. All INDI properties are correctly seen my PHD2 and can be controlled with INDI panel fine. Using bundled older INDI version works fine. I'm not 100% sure, but I think system INDI still worked fine around 1.9.9 or 1.9.8 as well.

Debug log attached, interesting part looks like:
```
21:47:57.924 00.001 139810526706944 gear_dialog: OnButtonConnectScope
21:47:57.924 00.000 139810526706944 Status Line: Connecting to Mount ...
21:47:57.925 00.001 139810526706944 Connecting to mount [INDI Mount [EQMod Mount]]
21:47:57.925 00.000 139810526706944 INDI Mount connecting to device [EQMod Mount]
21:47:58.426 00.501 139810459215424 INDI Mount: new device EQMod Mount
21:48:12.943 14.517 139810526706944 INDI Mount bg connection failed canceled=0
21:48:12.943 00.000 139810526706944 Alert: Cannot connect to mount EQMod Mount: Connection timed-out
21:48:12.950 00.007 139810526706944 INDI Mount: serverDisconnected
21:48:12.950 00.000 139810526706944 Error thrown from /home/jpaana/src/phd2/scope_indi.cpp:947->INDI Scope: cannot get side of pier when not connected
21:48:12.950 00.000 139810526706944 INDI Mount: SideOfPier returns -1
21:48:12.950 00.000 139810526706944 INDI Mount: connectServer done ready = 0
21:48:12.951 00.001 139810526706944 Throw from /home/jpaana/src/phd2/gear_dialog.cpp:1403->OnButtonConnectScope: connect failed
21:48:12.951 00.000 139810526706944 Status Line: Mount Connect Failed
21:48:13.553 00.602 139810526706944 gear_dialog: OnButtonConnectAuxScope
21:48:13.553 00.000 139810526706944 Status Line: Connecting to Aux Mount ...
21:48:13.572 00.019 139810526706944 Connecting to aux mount [INDI Mount [EQMod Mount]]
21:48:13.572 00.000 139810526706944 INDI Mount connecting to device [EQMod Mount]
21:48:13.884 00.312 139809976743488 INDI Mount: new device EQMod Mount
21:48:28.598 14.714 139810526706944 INDI Mount bg connection failed canceled=0
21:48:28.598 00.000 139810526706944 Alert: Cannot connect to mount EQMod Mount: Connection timed-out
21:48:28.603 00.005 139810526706944 INDI Mount: serverDisconnected
21:48:28.603 00.000 139810526706944 Error thrown from /home/jpaana/src/phd2/scope_indi.cpp:947->INDI Scope: cannot get side of pier when not connected
21:48:28.603 00.000 139810526706944 INDI Mount: SideOfPier returns -1
21:48:28.603 00.000 139810526706944 INDI Mount: connectServer done ready = 0
21:48:28.603 00.000 139810526706944 Throw from /home/jpaana/src/phd2/gear_dialog.cpp:1454->OnButtonConnectAuxScope: connect failed
21:48:28.603 00.000 139810526706944 Status Line: Aux Mount Connect Failed
```

[PHD2_DebugLog_2023-02-06_214743.txt](https://github.com/OpenPHDGuiding/phd2/files/10678368/PHD2_DebugLog_2023-02-06_214743.txt)

pawel-soja, 2023-02-07:
Okay, I think I know what happened.
Recent changes are fine [Add compatibility with INDI 2.0 and cleanup #1033](https://github.com/OpenPHDGuiding/phd2/pull/1033).
No need to remove wrapper in pull request [Report error if INDI v2 is being used #1035](https://github.com/OpenPHDGuiding/phd2/pull/1035).

The problem when connecting to an INDI server appeared in version 1.9.9, where the request to get properties is made after the connection information:

[Source](https://github.com/indilib/indi/blob/8ec6fd7001fd9a2c5ab8cc0c5c1a5add1b9afcc1/libs/indiclient/baseclient.cpp#L331-L335):
```cpp
    serverConnected(); // call virtual function

    d->userIoGetProperties(); // get device properties
```

The implementation of the `serverConnected` function in PHD2 waits for a device and this causes a deadlock.

I think I will make fixes in INDI Core, but I am thinking about a solution in PHD2, which will also be compatible for now.

pawel-soja, 2023-02-07:
Fixed: [Fix serverConnected call too early for INDI Core 1.9.9/2.0.0 #1037](https://github.com/OpenPHDGuiding/phd2/pull/1037)

jpaana, 2023-02-07:
Thanks, can confirm that PR fixes my issue.

agalasso, 2023-02-19:
fixed by #1037

closed 2023-02-19 by agalasso