OpenPHDGuiding/phd2 #1038: build fails with override error

justanenduser, 2023-02-12:
Fully updated 64 bit  Fedora 37. Fully updated phd2 (pulled from git 02/11/2023). 

[ 48%] Building CXX object CMakeFiles/phd2.dir/scope_indi.cpp.o
In file included from /home/gary/Downloads/phd2/config_indi.h:43,
                 from /home/gary/Downloads/phd2/scope_indi.cpp:41:
/home/gary/Downloads/phd2/phdindiclient.h:49:10: error: ‘bool PhdIndiClient::connectServer()’ marked ‘override’, but does not override
   49 |     bool connectServer() override;
      |          ^~~~~~~~~~~~~

imaluschi, 2023-02-19:
Same here, Kubuntu 22.10, pulled 02/18/2023

[image]

imaluschi, 2023-02-19:
Ok, got it. With hint from #1032 use -DUSE_SYSTEM_LIBINDI=ON with cmake.
Then this error is gone
(I'm now running into another problem, " /usr/bin/ld: /usr/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so: undefined reference to `TIFFGetVersion@LIBTIFF_4.0' " and other TIFF's ).

pchev, 2023-02-19:
I also get the same with the INDI 1.8.7 included with the PHD2 source code.

@pawel-soja  what are the INDI version supported by the connectServer() change?  
It work fine with INDI 2.0.0,  but I see this need many change to the PHD2 thirdparty.cmake.

pawel-soja, 2023-02-19:
The function has been marked as virtual since version 1.9.9, so I added `override` too quickly.
Thanks @pchev for letting me know about this issue.
I'll prepare a fix soon.

pawel-soja, 2023-02-19:
For some specific reason, GitHub Actions are not running here?

agalasso, 2023-02-19:
fixed by #1040

closed 2023-02-19 by agalasso