OpenPHDGuiding/phd2 #1041: Add indi rotator support

mgutierrezp, 2023-03-01:
Hi,

I would please request indi rotators support for phd2 (linux version, obviously). Currently it seems is not supported. I have an electronic rotator (falcon) before my oag. Any rotation forces me to recalibrate. This is a major problem since I'm running an automated setup during the night and imaging multiple objects with different rotation.

Regards

mgutierrezp, 2023-03-07:
dear developers; it's not my intention to push you, honestly. I would only like to know if it is feasible to implement this feature, just to know and plan to switch to ascom :(

pchev, 2023-03-07:
I just take a quick look at the existing rotator code and INDI rotator simulator properties and I think this is possible to add to PHD2.
The equivalent of ASCOM Position is ABS_ROTATOR_ANGLE.ANGLE.
But unfortunately I not have the time to work on that right now.

d33psky, 2023-03-07:
I'm in the same boat as Patrick.
One of the INDI drivers I wrote was a rotator+focuser and I've contributed some Linux code to PHD2 but I have just no time to work on your request right now.

mgutierrezp, 2023-05-14:
> Thanks to all of you guys for looking into this. I fully understand you
> are time-limited and working selflessly in this project. All phd2 users
> thank you for your job and effort.
>
> best

Dear developers,

I was just wondering if you have had any chance to get into this. Really
appreciate your effort.

Regards

comptonizing, 2023-08-26:
I will be facing the same situation soon, so I added the rotator support, see #1091 .

mgutierrezp, 2023-08-26:
Philipp, you made my day mate. I've just compiled your branch and works
flawlessly. Guiding is horrible cause it's very windy here, but the rotator
is recognized perfectly and the guiding still works and corrections seem
consistent after some rotations.
I'm very grateful for this implementation. Thanks a lot. Really appreciate
it.

Regards

comptonizing, 2023-08-27:
No worries, very happy to have done something useful in return, after many years of just using the software.
Also, thanksa lot for testing it in a real scenario, I only had the simulators available.