OpenPHDGuiding/phd2 #1042: Can we clean up the file directory ?

AstroAir, 2023-03-04:
When I am looking for files, I am often troubled by too large root directory. Can we put a corresponding function in a folder. This can make the whole project more refreshing. I saw someone put forward the same idea seven years ago, but it was shelved because some very important functions were being made at that time. So can we restart this work now.
This may be a big task, but I believe we can all accomplish it. I hope my opinion can be adopted.

Here are my basic ideas:
```
/
    /algorithm
    /device
        /camera
        /telescope
    ...
```

agalasso, 2023-03-04:
@AstroAir this has been bothering me too for quite a while, and there's nothing stopping us from doing this now. Do you want to make a PR, or a series of PRs (incremental changes)? If you would like to proceed with this, let's discuss the hierarchy we'd like to see.
I would prefer to avoid a deep hierarchy, and we probably do not need an intermediate /device level.
How about this:
```
/         - README LICENSE etc
/src      - phd.cpp etc
/src/camera - cam_*
/src/scope - scope_*
/src/stepguider - stepguider_*
/src/rotator - rotator_*
/src/guide_algorithm - guide_algorithm_*
```
Third party files are also a bit of a mess. We have cameras/ thirdparty/ extra_frameworks/ -- all 3d party code. Cleaning that up can be a separate PR or PRs.

d33psky, 2023-03-04:
I quite like the proposals here !

AstroAir, 2023-03-06:
@agalasso After sorting out the directory, all the translation files are invalid, which seems to be a big project, which bothers every sorting. How should we solve this problem?

agalasso, 2023-03-06:
Agreed, this will not be a small/easy project. Besides those fits files you identified (apparently committed inadvertently to the repo), I do not think we have many unnecessary or unused files.
If you still want to work on this, I would suggest an incremental approach - a series of complete (tested and reviewed) smaller prs that can be merged sequentially:
 1. remove unused files
 1. make a src directory and move all the source files into there
 1. make device-specific directories (camera, scope, stepguider, rotator)
 1. ... etc. ...

At each step we would want to build and test on all 4 platforms (Windows, Mac 64, Mac 32, Linux).
This will allow us to make forward progress while keeping the changes minimal at each step so that we can more easily review and discuss the changes.
The alternative approach of doing everything all at once in a big pr would reduce the testing effort, but it will be more difficult to get consensus on where the files belong and keeping the pr in sync with master while questions are worked out will be very difficult if the pr drags on for a long time.

closed 2024-06-18 by AstroAir