OpenPHDGuiding/phd2 #1055: Build is failing on Raspberr Pi OS 64bit

phomer60, 2023-04-01:
[ 70%] Building CXX object CMakeFiles/phd2.dir/calibration_assistant.cpp.o
/home/indi/Projects/phd2/calibration_assistant.cpp: In function ‘wxString VectorToString(const wxString&, const std::vector<wxString>&)’:
/home/indi/Projects/phd2/calibration_assistant.cpp:789:10: error: ‘for_each’ is not a member of ‘std’
  789 |     std::for_each(vec.begin(), vec.end(), [&l](const wxString& s) { l += s.length(); });
      |          ^~~~~~~~
/home/indi/Projects/phd2/calibration_assistant.cpp:792:10: error: ‘for_each’ is not a member of ‘std’
  792 |     std::for_each(vec.begin(), vec.end(),
      |          ^~~~~~~~
make[2]: *** [CMakeFiles/phd2.dir/build.make:1233: CMakeFiles/phd2.dir/calibration_assistant.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:971: CMakeFiles/phd2.dir/all] Error 2
make: *** [Makefile:182: all] Error 2
indi@HOM-INDI-BE-3:~/Projects/build/phd2 $

pchev, 2023-04-01:
Same problem for me with gcc 12.2 on Ubuntu 22.10.

The solution is to add #include <algorithm> in calibration_assistant.cpp after line 38.
@agalasso and @bwdev01 , can you look if this is the good solution and if this also work on Windows and Mac?

phomer60, 2023-04-01:
I can confirm that the suggested solution worked for the Raspberry Pi.

Thanks 

Paul

agalasso, 2023-04-01:
Thanks Patrick, that fix works on Windows.

PR #1058 (2023-04-02): fix compile error on Linux/Mac
fixes #1055
closed 2023-04-02 by agalasso