OpenPHDGuiding/phd2 #1056: PHD2 crash when build with the bundled INDI 1.8.7

pchev, 2023-04-01:
When build with the bundled indi-1.8.7.tar.gz , PHD2 crash after receiving the first camera image.
The crash is in indi-1.8.7/base64.c when accessing out[] in from64tobits_fast().

I try to replace INDI 1.8.7 by INDI 1.8.2 provided by Ubuntu 20.04 and it work without problem.

@paolostivanin  , any reason you select 1.8.7?  can it be replaced by 1.8.2?

paolostivanin, 2023-04-03:
hello @pchev ! It was a first step on updating the codebase to a more recent INDI version, but then RL got in the way...
Sorry to hear that this change broke something for you. I will create a PR to revert this specific thing.

pchev, 2023-04-04:
I make more test and find that INDI version 1.8.3 work, but since 1.8.4 it crash.

There is no interest to try to fix PHD2 for this old versions, the important point is to have a bundled version that work without issue and compile without warning.

I also tested with INDI 2.0.1 released today and it work fine, but compilation show a lot of deprecated warning. 
Future work must be to adapt to INDI 2.0 to fix this warning and after that we can bundle a version 2.0.x.

I create a PR with 1.8.3 now.

commit 5fa9c69c (2023-04-04): Replace INDI 1.8.7 by 1.8.3 to fix camera crash, fix #1056 (#1060)
closed 2023-04-04 by agalasso