OpenPHDGuiding/phd2 #1088: find_star method from event server have lower performances than its UI counterpart

gnthibault, 2023-07-23:
To whom it may concern.
I have been playing with the event server for quite some time now, but still experiencing issues with robust star selection method.
Basically I know where is the star I want to guide one (as part of a spectroscopic workflow), and I send a find_star command to the event server, which is documented here:
https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#available-methods

The signature seems to be: find_star(x,y,width,height)
Which I use with x, y being the leftmost, lowermost coordinates (in sensor coordinate system) of my bounding box (usually 50 pixels), and widht, height the respective size in pixel of the desired search bounding box.

Unofrtunately this method seems extremely inconsistent, end more importantly, it is much less robust that the autoselection that works perfectly in the UI, which I am testing quite often at the same time that I try to debug the actual event server method.

Unfortuantely the code is a bit cryptic to me, but in practice it seems that event server method is completely different from the UI method where you just have to click somewhere neat your star for it to be detected properly.
Can someone help me understand how to reach the same performances of detection from event server, as what is provided from the UI ?

Thank you in advance for your help 
Best regards

gnthibault, 2023-07-23:
Ok I confirm it seems that this dummy commit fully fixes the issue I am experiencing (but that's not likely what the project wants): https://github.com/gnthibault/phd2/commit/4691361d5a48cff3bc5af755c41c97aa9b220267

Not doing a PR at the moment because it wouldn't make sense as is. But I'll be interested in getting feedback about how to improve that to make a PR that would go through.

Thank you in advance for your help

bwdev01, 2023-07-24:
I don’t think you’ve given us enough information to work with here – just comments about inconsistency and lack of robustness in general terms which I don’t know how to interpret.   I think you should run a series of back-to-back comparisons of the interactive and event-server driven find operations on the same stars.  Note the times you did each pair, then submit the debug log file for that session, telling us the timestamps of interest and what results you are unhappy about.  Also, please move the discussion to our support forum and upload the full debug log file using the built-in PHD2 Upload Logs feature:

 

https://openphdguiding.org/getting-help/

 

If you look at the debug log file yourself, you may discover there are unintended differences between the two operations, e.g. the spatial region that is actually being searched.

 

Regards,

Bruce

 

https://github.com/OpenPHDGuiding/phd2/wiki/EventMonitoring#available-methods

bwdev01, 2024-03-14:
Unless we get raw log data that clearly demonstrates the problem, I plan to close this.

closed 2024-06-24 by agalasso