OpenPHDGuiding/phd2 #1090: PHD2 cannot handle mount and camera on different indi server

gnthibault, 2023-08-13:
To whom it may concern,
I am currently facing an issu with openPHD2. Whenever I click on the connection button to setup my profile, I notice that I need to select mount an camera device given an Indi server and port.
Unfortunately it appears that I cannot use one camera on one server, and one mount on another server.

Unfortunately, this is exactly what I need to do currently on my setup. I think this is more or less a bug, as one can enter a different value for indi server host in camera and mount, but later on the software crashes.
Supporting multiple parallel indi client would definitely be the way to go for me.

Thank you in advance for your help and support.

agalasso, 2023-08-13:
@gnthibault thanks for letting us know about the issue

pchev, 2023-08-13:
The way to do that with INDI is to link the server together as explained in the page:
https://www.indilib.org/support/tutorials/159-indi-on-multiple-devices.html

There is multiple advantage to that, specifically the INDI server can still fill the FITS header with the telescope information even it is connected to the other server.

To prevent multiple image transport on the network be sure you connect PHD2 to the server with the camera and create  the link to the telescope on this server.

gnthibault, 2023-08-13:
Thank you very much for the feedback @pchev .
This will make my workflow a bit more complicated as it introduces a dependency between servers, and I cannot reset servers in an arbitrary order, but I will need to stop and restart individual drivers on the master server in the right order. But I will try that.

gnthibault, 2023-08-13:
Sorry I changed my mind, I think there are still good reason for having phd2 to manage more than one client. Linking indi server is a good workaround but still a workaround from my point of view (I tag fits with observatory informations with my own application for instance)