OpenPHDGuiding/phd2 #1123: raspberry pi ASI120mm mini not detected on 2.6.11dev6 64bit, works on 2.6.11 release 32 bit

Akinetopsia1, 2023-11-20:
Raspberry 4

i compiled the above dev snapshot when i upgraded from the outdated 32bit astroberry distro but i can't get my asi120 mini to be detected on my new debian bullseye 64bit install. 

It works without issues if i switch back to my previous sd card. Camera is on the same port.

[image]

Akinetopsia1, 2023-11-21:
I found the debug log.

Here's what i believe are the relevant parts, but i attached the clean log (fresh open app, click connect, quit app)

`16:20:26.312 00.001 547737441536 GetString("/profile/3/camera/LastMenuChoice", "None") returns "ZWO ASI Camera"
16:20:26.312 00.000 547737441536 CameraFactory(ZWO ASI Camera)
16:20:26.312 00.000 547737441536 GetBoolean("/profile/3/camera/UseSubframes", 0) returns 0
16:20:26.312 00.000 547737441536 GetInt("/profile/3/camera/ReadDelay", 150) returns 150
16:20:26.313 00.001 547737441536 GetInt("/profile/3/camera/gain", 95) returns 95
16:20:26.313 00.000 547737441536 GetInt("/profile/3/camera/TimeoutMs", 15000) returns 15000
16:20:26.313 00.000 547737441536 GetInt("/profile/3/camera/SaturationADU", 0) returns 0
16:20:26.313 00.000 547737441536 GetBoolean("/profile/3/camera/SaturationByADU", 1) returns 1
16:20:26.313 00.000 547737441536 GetDouble("/profile/3/camera/pixelsize", 0.000000) returns 0.000000
16:20:26.313 00.000 547737441536 GetInt("/profile/3/camera/binning", 1) returns 1
16:20:26.313 00.000 547737441536 GetInt("/profile/3/camera/ZWO/bpp", 8) returns 16
16:20:26.313 00.000 547737441536 Created new camera of type ZWO ASI Camera = 0x55a47a4250
16:20:26.313 00.000 547737441536 GetString("/profile/3/camera/LastMenuChoice", "") returns "ZWO ASI Camera"

[...]

16:20:26.674 00.343 547737441536 starting server
16:20:26.675 00.001 547737441536 event server started, listening on port 4400
16:20:26.675 00.000 547737441536 Server started, listening on port 4300
16:20:26.676 00.001 547737441536 Status Line: Server started

[...]

16:20:41.643 14.922 547737441536 gear_dialog: OnButtonConnectAll
16:20:41.643 00.000 547737441536 gear_dialog: DoConnectCamera [ZWO ASI Camera]
16:20:41.644 00.001 547737441536 Status Line: Connecting to Camera ...
16:20:41.649 00.005 547737441536 GetString("/profile/3/cam_hash/6ff84bf4b36abbb1/whichCamera", "") returns ""
16:20:41.650 00.001 547737441536 Connecting to camera [ZWO ASI Camera] id = []
16:20:41.650 00.000 547737441536 ZWO: SDK Version = [1, 30, 0, 0]
16:20:41.655 00.005 547737441536 ZWO: find camera id: [], ncams = 1
16:20:41.659 00.004 547737441536 ASIGetCameraProperty ret 5
16:20:41.660 00.001 547737441536 Alert: Failed to get camera properties for ZWO ASI Camera.
16:20:41.679 00.019 547737441536 Throw from /home/astropi/source/phd2/phd2/gear_dialog.cpp:1093->DoConnectCamera: connect failed
16:20:41.680 00.001 547737441536 Status Line: Camera Connect Failed`

[PHD2_DebugLog_2023-11-21_162023.txt](https://github.com/OpenPHDGuiding/phd2/files/13431150/PHD2_DebugLog_2023-11-21_162023.txt)

Camera is connected and detected by the OS.

[image]

Akinetopsia1, 2023-11-22:
FIXED.

I wrongly assumed the ASI sdk/driver files would be compiled as well as PHD2, but I fixed it by manually installing the ZWO ASI SDK after compiling PHD2, using the following:

$ sudo bash -c 'echo "deb [trusted=yes] https://apt.fury.io/jgottula/ /" > /etc/apt/sources.list.d/jgottula.list'
$ sudo apt update
$ sudo apt install libasicamera2
$ sudo reboot

Camera is now detected and working properly.

agalasso, 2023-11-22:
Thanks for reporting back, glad to hear you got it working!

closed 2023-11-22 by agalasso

Akinetopsia1, 2023-11-22:
@agalasso thanks for the awesome bit of software!

PHD provides precompiled libasicam2 binaries in **cameras/zwolibs/**, correct ? I haven't investigated much further in what the issue was specifically but it may be that the provided binaries either are not the in right executable format or missing the rpi 64 specific architecture

I found this random deb repo with installable precompiled zwo packages but i'm unsure if there is an official one. Trying to download the sdk directly from zow's website only provided general linux precompiled archive instead of the source code, which probably would have given the same result.