OpenPHDGuiding/phd2 #1125: v2.6.12 Camera tab messed up in advanced settings

samhaa, 2023-11-26:
v2.6.12 Camera tab messed up in advanced settings when camera has been connected.

[image]

agalasso, 2023-11-26:
@samhaa thanks for letting us know about the issue.
Could you please provide more information:
 - what os and version are you using (Windows? Linux? If Windows, what version -- Windows 10, Windows 11, something else?)
 - What camera are you using?
 - What are the exact steps that you do in PHD2 to get the problem to show up?  For example: open, phd2, click the connect equipment button, than what?

Could you please provide a link to your PHD2 logs -- the log file upload instructions are here: https://openphdguiding.org/getting-help/

bwdev01, 2023-11-26:
Also, you should be sure you are running the latest version of your camera drivers.  There have been many changes to the vendor-supplied SDK's during the life of 2.6.11.

PR #1130 (2023-12-05): Fixes for camera tab of Advanced Settings dialog
Reduce nesting of sizers in AD camera tab and do not rely on "Has" properties to create cam property controls.

Fixes #1125
commit 3e73fab4 (2023-12-05): Reduce nesting of sizers in AD camera tab, don't rely on "Has" properties to create cam property controls. Fixes #1125 (#1130)
closed 2023-12-05 by bwdev01