OpenPHDGuiding/phd2 #1136: Building on Raspberry Pi OS bookworm failing

phomer60, 2023-12-19:
Do we need updated prerequisites for bookworm?

sudo apt-get install build-essential git cmake pkg-config libwxgtk3.0-gtk3-dev \
   wx-common wx3.0-i18n libindi-dev libnova-dev gettext zlib1g-dev libx11-dev \
   libcurl4-gnutls-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package wx3.0-i18n is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package libwxgtk3.0-gtk3-dev
E: Couldn't find any package by glob 'libwxgtk3.0-gtk3-dev'
E: Couldn't find any package by regex 'libwxgtk3.0-gtk3-dev'
E: Package 'wx3.0-i18n' has no installation candidate

jpaana, 2023-12-23:
Looks like Bookworm uses libwxgtk3.2-dev instead.

phomer60, 2023-12-24:
Thanks, that has solved the issue, it probably needs someone to update the build notes.

Paul

agalasso, 2023-12-26:
I updated the instructions on the Wiki page -- closing this issue now.

closed 2023-12-26 by agalasso