OpenPHDGuiding/phd2 #1152: PHD2 no more build in Ubuntu PPA

pchev, 2023-12-27:
After the last changes in the PHD2 build system the build failed for the Ubuntu PPA at 
https://launchpad.net/~pch/+archive/ubuntu/phd2 

I already make two change I will add to a PR later, for now this is in branch patrick/fix-ppa

- Because the PPA build server cannot access external network, fetching INDI from Github do not work.   
  I make this change to build with system libindi: https://github.com/OpenPHDGuiding/phd2/commit/c5d84f5a6d500523c451a563694422989630df22
  And add a dependency to ppa:mutlaqja/ppa because no Ubuntu version package INDI 2.0 for now.

- Build with system libindi fail in cmake, this is the fix:  https://github.com/OpenPHDGuiding/phd2/commit/dec3c2c6bedd0885cfb34d1ef0c43b8c57c3ee7a

Now the build fail because of gtest, probably because of this change : https://github.com/OpenPHDGuiding/phd2/commit/94cafff72ca4a026f0a80bd786734355f46d7b76

First the new requirement for cmake 3.24 is not available in current stable Ubuntu 22.04, the error message is:
CMake Error at CMakeLists.txt:31 (cmake_minimum_required):
  CMake 3.24 or higher is required.  You are running version 3.22.1

This version of cmake is available in Ubuntu 23.10 but then it fail when compiling with gtest:
/<<PKGBUILDDIR>>/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:38:10: fatal error: gtest/gtest.h: No such file or directory
   38 | #include <gtest/gtest.h>

The full log is at : https://launchpadlibrarian.net/706009692/buildlog_ubuntu-mantic-amd64.phd2_2.6.13.rev20231227d-0ppa1~ubuntu23.10_BUILDING.txt.gz

I not really understand how gtest is installed but if this is from the network we can reach the same limitation as for INDI.
All work fine when compiling on my local computer.
Please @agalasso can you take a look at this issue.

agalasso, 2023-12-27:
Thanks @pchev I will take a look at the issues.

pchev, 2023-12-27:
I just look more in detail at https://github.com/OpenPHDGuiding/phd2/commit/94cafff72ca4a026f0a80bd786734355f46d7b76 and remark it suppress the option USE_SYSTEM_GTEST.
There is a libgtest-dev version 1.13 packaged with Ubuntu, maybe the more simple is to revive this option.

agalasso, 2023-12-27:
yes, reinstating USE_SYSTEM_GTEST is a good solution; the other thing I was thinking would be to make building the test executables optional and perhaps disabled by default. gtest is only needed for building the test executables, and those are not needed for the debian package.

pchev, 2023-12-27:
It is good if we can keep the test, because the test are run as part of the build process and the build fail in case of error in the test.
This can detect an issue with the Eigen library like we have some years ago with the i386 version.

agalasso, 2023-12-27:
ah, that's great that the test is run as part of the ppa build!  USE_SYSTEM_GTEST sounds like the right solution then.  I'll put up a pr for that directly against master

agalasso, 2023-12-27:
#1153 should address the gtest issue

looking into fixing the cmake_minimum_required issue now...

agalasso, 2023-12-27:
@pchev I pushed another commit to #1153 's branch which allowed building on focal (cmake 3.16)

PR #1155 (2023-12-27): Patrick/fix ppa
Now everything build fine on all supported Ubuntu version.

This PR is for the two modified Debian files to use the new build option, thank you very much for that.

This fix #1152
closed 2023-12-27 by agalasso