OpenPHDGuiding/phd2 #1157: Using Open Astro Guider (IMX290) with PHD2

peteasa, 2023-12-29:
I will come up with some proposed patches to the code at some point but for now 
ISSUE / FIX1 : here is a suggestion that I mentioned in https://groups.google.com/g/open-phd-guiding/c/vVk9Qa6H7XI/m/QAJWXJ1XAAAJ - to my mind this is very generic and might apply to any colour web cam

The problem "Unsupported type or read error loading FITS file" reported and no camera connection can be made to PHD2.

looking at phd2/cam_indi.cpp .. the fits file produced by Open Astro Guider and INDI V4L2 CCD driver for the IMX290 chip has NAXIS = 3 / number of data axes .. so the phd2 code can't handle this type of image.   A simple modification to phd2/cam_indi.cpp fixes this:
`-    if (nhdus != 1 || naxis != 2)
+    if (nhdus < 1 || naxis < 1)
     {
         pFrame->Alert(_("Unsupported type or read error loading FITS file"));
         PHD_fits_close_file(fptr);
`
I do not know what other implications this change has but it did allow me to get the camera looping to work in PHD2.

ISSUE / POSSIBLE_FIX2 : Apparently there are two meridian options.. One measuring angles from 0 degrees (I am based in London so this is what I would expect) and the other measuring angles from 180 degrees (I guess is I lived in Hawaii that is what I would expect).  This causes a bit of a problem with the Open Astro Guider... see https://wiki.openastrotech.com/en/Knowledge/Software#useful-python-scripts 

> there is some mismatch between Meade protocol implementation in generic LX200 GPS INDI driver for Linux and OAT ASCOM driver for Windows

> OAT expects 179W is 359 and 179E is 1 from 180 at Greenwich.... LX200 sends 179W is -179 and 179E is 179 from 0 at Greenwich

I am working on this second issue at the moment.. 
My immediate thought was to use Meridian offset (degrees) = -179 ..https://openphdguiding.org/man/Tools.htm#Calibration_Assistant
However this number is restricted to a maximum of 50 degrees.  The result of this is when I press the Slew button on the calibration assistant page the mount slews to the North and attempts to Az 180 DE 0 that is actually pointing into the ground.

The problem "Can't press the SLEW button on  the calibration assistant page without the Open Astro Tracker crashing into its supports"

My POSSIBLE_FIX2 is to modify Meridian offset (degrees) to allow setting 180 degrees to handl this apparent mismatch.

Open to any comments suggestions / other alternative configurations.

pchev, 2023-12-29:
Fix1 is not  necessary, just open the INDI panel for the V4L2 camera and at the Image Settings tab check Format = Mono instead of RGB.

About the second issue, there was a bug with LST for INDI mount that was fixed some time ago, be sure to use the last available version of PHD2.

agalasso, 2023-12-29:
@peteasa thank you for reporting those issues.  I'm going to close this ticket based on @pchev 's response.  If those solutions do not work for you, please open another github issue -- preferably one issue per problem (LST, FITS, ...) so we can track the issues independently.

peteasa, 2023-12-30:
@agalasso thank you for your responses.  The issues remain as described.. closing the report adds no value to this discussion!

The problem "Unsupported type or read error loading FITS file" reported and no camera connection can be made to PHD2.

In what dialect of English does this error message mean: "Open the INDI panel for the V4L2 camera and at the Image Settings tab check Format = Mono instead of RGB"

@pchev - you have nothing to fear about making this change.  When you run calibration with the change it makes the recommendation to stack greyscale images to improve calibration.  This a far better outcome than being left thinking that the FITS file is invalid.  Either the code has to change or the error message has to change or I leave it up to you to add a description in the very good documentation that comes with PHD2 to explain why "Unsupported type or read error loading FITS file" means what you say it means.

I am using the latest PHD2 code - https://github.com/OpenPHDGuiding/phd2/blob/3c41fb20621e26cdd861747db47d156d08ccb354/cam_indi.cpp#L862

On my second comment this issue is as described - The problem "Can't press the SLEW button on the calibration assistant page without the Open Astro Tracker crashing into its supports"

The two issues remain as reported.  In my view this issue has not been addressed satisfactorily.

pchev, 2023-12-30:
I thought you were fluent with the INDI settings, hence the condensed instructions I'm giving. But if you not understand my English we can continue in French.

Maybe a screenshot is more clear:
[image]

Please ensure your setting is the same as in the screenshot and try to loop image from the camera.

I really not think a code change is required. Thousand of people successfully use webcam for guiding for years.
About your proposed change, naxis < 1 is invalid because we cannot detect stars in a one dimension image.
Making a changed to accept a 3 dimension image is not optimal. In this case only the first color channel is used, generally the red color. This mean we use only one pixel on four resulting in degraded star signal.

As indicated by Andy please open another issue for the meridian problem.

peteasa, 2023-12-30:
@pchev thanks for the info.. and no need to reply... this message is simply to **close the issue** from my side...

I will consider opening a separate issue per fix once I have the whole system working and I understand more.  At the moment I am trying to debug why the phd2 tracking fails for my setup.  I think its likely to do with power surges from the drive motors crashing the USB 3 interface...

Thanks for the further information about the reason for not averaging the three channels of data when colour images are provided... (another possible option for fixing the multiple channel support)..

Note that I am not the only one who reports this problem with the web cam connections.. a just found this 2020 chat - https://groups.google.com/g/open-phd-guiding/c/xc1T6KQVXMs with exactly the same problem and solution.

Clearly from what you say you consider the error "Unsupported type or read error loading FITS file" to be entirely satisfactory and clear to anyone who reads it.  So you must then be happy to receive / discuss the issue with the next poor chap who stumbles upon the same error.

Happy new year!

pchev, 2023-12-30:
Averaging the 3 channel is a bad option for performance, this is extra work for the INDI driver to split the channel, then extra work for PHD2 to recombine.

I agree the message can be better, maybe add a first test for (naxis == 3)  that give the message "RGB images are not supported, please switch to Mono", after that do the current test for the other case.

On the other subject I read more of the documentation of the OAT. 
As you indicate in the first message it look like it is not fully compatible with the INDI driver. 
The FAQ "OAT slews into a weird location in INDI/KStars" indicate a similar issue to the one you get with the slewing from the calibration assistant. The given solution to try different timezone until it work is not very serious and probably need a fix in the driver.

PR #1159 (2023-12-30): Better error message for RGB FITS file from INDI
Change the error message when receiving a FITS file from INDI with NAXIS=3
Please comment if the new message is not clear enough.

Fix #1157
closed 2023-12-30 by agalasso