OpenPHDGuiding/phd2 #1161: Add a calculator for worm gear ratio

apos, 2024-01-21:
Hi there,

your table is very interesting. BUT, it is very easy to calculate the ratio, if you know the teeth of your RA/DEC gear:

See: https://www.cloudynights.com/topic/806044-eq6r-pro-worm-cycle/?p=11619011

> Assuming the mount has true sidereal tracking the worm wheel rotates exactly once in a full sidereal day (86,164.1 seconds).
> 
> Obviously the worm wheel must have a whole number of teeth (180, 176, 160, etc.). The division is then simplified  (86164.1/# of Teeth).
> 
> If the worm wheel has 180 teeth then the worm gear rotates once in 478.6894 seconds. (Ignoring backlash or periodic error)

You should simply add this to your readme page ;-)

Cheers, Axel

bwdev01, 2024-01-21:
This doesn't have anything to do with PHD2, you probably posted to the wrong place.

closed 2024-01-21 by bwdev01

apos, 2024-01-21:
Absolute certainly not @bwdev01 - sorry :-)

Multistar guiding is a key feature not just of PHD2. Therefore you need to know (and calculate) the worm period and tell this the software.

This is, what this Readme ist about: https://github.com/OpenPHDGuiding/phd2/wiki/Mount-Worm-Period-Info

Correct me, if I am wrong (I am sure, I am not):

Cheers, Axel

apos, 2024-01-21:
@bwdev01 ... also a way of closing tickets ...

bwdev01, 2024-01-21:
Sorry, I’m not used to seeing feature requests for the Wiki.  I think you can edit the Wiki yourself but if not, you can do a PR.  I find it hard to imagine we have many people out there who know gear teeth counts but not worm periods, but who knows.  The information is only there to help people provide an initial period length for the PPEC algorithm which is optional in any case.  Beyond that, the worm period has nothing at all to do with multi-star guiding. 

 

Bruce