OpenPHDGuiding/phd2 #1173: macos crash rendering bitmap

agalasso, 2024-02-12:
From a user in the forum:

```
This is the crash report. There is no log. The application crashes as soon as you try to connect a camera.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PHD2 [3021]
Path:                  /Applications/PHD2.app/Contents/MacOS/PHD2
Identifier:            PHD2
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-02-12 08:55:09.7122 -0800
OS Version:            macOS 14.3.1 (23D60)
Report Version:        12
Anonymous UUID:        83B90DC4-8BB1-F41E-5E1F-5F3A412327BD

Sleep/Wake UUID:       6E8F0D8B-A1F1-4A72-AC7F-7176248D97F9

Time Awake Since Boot: 4200 seconds
Time Since Wake:       272 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x100896B24)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [3021]

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x100896B24)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   PHD2                                     0x100896b24 __exp10 + 4
1   PHD2                                     0x100705413 nsvg__atof(char const*) + 227
2   PHD2                                     0x100704978 nsvg__parseCoordinateRaw(char const*) + 216
3   PHD2                                     0x100704b6b nsvg__parseCoordinate(NSVGparser*, char const*, float, float) + 27
4   PHD2                                     0x100703dd4 nsvg__parseAttr(NSVGparser*, char const*, char const*) + 2036
5   PHD2                                     0x10070251c nsvg__parseGradientStop(NSVGparser*, char const**) + 92
6   PHD2                                     0x1006fa7d4 nsvg__parseXML(char*, void (*)(void*, char const*, char const**), void (*)(void*, char const*), void (*)(void*, char const*), void*) + 772
7   PHD2                                     0x1006fa912 nsvgParse + 210
8   PHD2                                     0x100701d7b wxBitmapBundle::FromSVG(char*, wxSize const&) + 43
9   PHD2                                     0x100701f1b wxBitmapBundle::FromSVG(unsigned char const*, unsigned long, wxSize const&) + 75
10  PHD2                                     0x1005aa8c3 (anonymous namespace)::wxTangoArtProvider::CreateBitmapBundle(wxString const&, wxString const&, wxSize const&) + 323
11  PHD2                                     0x1005a84cc wxArtProvider::GetBitmapBundle(wxString const&, wxString const&, wxSize const&) + 476
12  PHD2                                     0x1006836ab wxInfoBarGeneric::ShowMessage(wxString const&, int) + 91
13  PHD2                                     0x10025c6bd MyFrame::DoAlert(alert_params const&) + 1645
14  PHD2                                     0x10025c8d6 MyFrame::Alert(wxString const&, void (*)(long), wxString const&, void (*)(long), long, bool, int) + 214
15  PHD2                                     0x10025b80e MyFrame::Alert(wxString const&, int) + 238
16  PHD2                                     0x100262538 MyFrame::DarkLibExists(int, bool) + 904
17  PHD2                                     0x1002626c4 MyFrame::SetDarkMenuState() + 36
18  PHD2                                     0x1001cbf63 GearDialog::DoConnectCamera(bool) + 6755
19  PHD2                                     0x1001ba9de GearDialog::OnButtonConnectCamera(wxCommandEvent&) + 110
20  PHD2                                     0x1007f8ec3 wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 515
21  PHD2                                     0x1007fa379 wxEvtHandler::ProcessEventLocally(wxEvent&) + 89
22  PHD2                                     0x1007fa1d8 wxEvtHandler::ProcessEvent(wxEvent&) + 40
23  PHD2                                     0x10066ecfc wxWindowBase::TryAfter(wxEvent&) + 204
24  PHD2                                     0x1007fa85c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
25  PHD2                                     0x100534904 wxToggleButton::OSXHandleClicked(double) + 132
[... 50 lines cut ...]
76  PHD2                                     0x10070d103 wxAppConsoleBase::MainLoop() + 99
77  PHD2                                     0x100539d7a wxApp::OnRun() + 26
78  PHD2                                     0x10076de36 wxEntry(int&, wchar_t**) + 86
79  PHD2                                     0x100288140 main + 48
80  dyld                                     0x201315386 start + 1942

On Monday, February 12, 2024 at 6:14:42 AM UTC-8 Bryan wrote:

    Please provide your guide AND debug logs per

    Welcome to the  [Open PHD Guiding](http://openphdguiding.org/) (PHD2) forum, telescope guiding software for astro-photographers.
    To get help with PHD2, please upload your log files - here's how: [How to ask for help with PHD2](http://openphdguiding.org/getting-help/).

    Bryan

    On Sunday, February 11, 2024 at 10:32:07 PM UTC-7 fhee...@comcast.net wrote:

        I have used PHD2 for over a year with various mounts, scopes, and arrangements  (including off axis, etc.) What happened now is that I downloaded the latest version (2.6.13)  on Friday and it crashes when I try to connect my QHY camera.

         

        I am running Sonoma 14.3.1 on an M1 MacBook Pro. All previous versions worked fine.

        Not sure how to proceed.
```

bwdev01, 2024-02-16:
Other reports are with debug logs are here:
[https://groups.google.com/g/open-phd-guiding/c/P8bLyrVrWfc/m/CChEH585BAAJ?utm_medium=email&utm_source=footer](url)

In the cases I've seen, the situation is being triggered by trying to display an alert message.

agalasso, 2024-03-09:
fixed by #1179

closed 2024-03-09 by agalasso