OpenPHDGuiding/phd2 #1202: cmake is failing on Raspberry Pi OS Bookworm unable to find OpenCV

phomer60, 2024-04-21:
Any ideas on how t resolve this?

cmake ~/Projects/phd2/
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- Performing Test HAS_CXX11_FLAG
-- Performing Test HAS_CXX11_FLAG - Success
-- Performing Test HAS_CXX0X_FLAG
-- Performing Test HAS_CXX0X_FLAG - Success
-- PHD2 version '2.6.13' (extracted from source)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- found version string 4.2
-- Found CFITSIO 4.2: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Using system's CFITSIO.
-- Found CURL: /usr/lib/aarch64-linux-gnu/libcurl.so (found version "7.88.1")  
-- using libcurl /usr/lib/aarch64-linux-gnu/libcurl.so
-- Using system's Eigen3.
-- Found Python3: /usr/bin/python3 (found version "3.11.2") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found wxWidgets: -L/usr/lib/aarch64-linux-gnu;-pthread;;;-lwx_gtk3u_aui-3.2;-lwx_gtk3u_core-3.2;-lwx_baseu-3.2;-lwx_gtk3u_html-3.2;-lwx_baseu_net-3.2 (found version "3.2.2") 
-- Found ASICamera2 lib /home/indi/Projects/phd2/cameras/zwolibs/armv8/libASICamera2.a
-- Found toupcam lib /home/indi/Projects/phd2/cameras/toupcam/linux/arm64/libtoupcam.so
-- Found ogmacam lib /home/indi/Projects/build/phd2/_deps/ogmacamsdk-src/linux/arm64/libogmacam.so
-- Found SVBCameraSDK lib /home/indi/Projects/phd2/cameras/svblibs/linux/armv8/libSVBCameraSDK.so
-- 'xgettext' program found at '/usr/bin/xgettext'
-- 'msgfmt' program found at '/usr/bin/msgfmt'
-- 'msgmerge' program found at '/usr/bin/msgmerge'
-- [DOC] Creating HTML target for locale en_EN
-- [DOC] Creating translation target for locale ar_LY
-- [DOC] Creating translation target for locale ca_ES
-- [DOC] Creating translation target for locale cs_CZ
-- [DOC] Creating translation target for locale de_DE
-- [DOC] Creating translation target for locale es_ES
-- [DOC] Creating HTML target for locale fr_FR
-- [DOC] Creating translation target for locale fr_FR
-- [DOC] Creating translation target for locale gl_ES
-- [DOC] Creating translation target for locale it_IT
-- [DOC] Creating HTML target for locale ja_JP
-- [DOC] Creating translation target for locale ja_JP
-- [DOC] Creating translation target for locale ko_KR
-- [DOC] Creating translation target for locale nb_NO
-- [DOC] Creating translation target for locale pl_PL
-- [DOC] Creating translation target for locale pt_BR
-- [DOC] Creating translation target for locale ro_RO
-- [DOC] Creating HTML target for locale ru_RU
-- [DOC] Creating translation target for locale ru_RU
-- [DOC] Creating translation target for locale uk_UA
-- [DOC] Creating translation target for locale zh_CN
-- [DOC] Creating HTML target for locale zh_TW
-- [DOC] Creating translation target for locale zh_TW
CMake Error at CMakeLists.txt:786 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/indi/Projects/build/phd2/CMakeFiles/CMakeOutput.log".

agalasso, 2024-04-22:
@phomer60 sorry for the trouble. Try `apt-get install libopencv-dev`.  Please let me know if that solves the issue for you and I'll update the Linux build instruction sin the wiki.

phomer60, 2024-04-22:
Andy, thanks, that solved the problem it is now building.

Paul

phomer60, 2024-04-22:
Build completed successfully.

closed 2024-04-22 by phomer60

agalasso, 2024-04-22:
thanks for the update, I updated the wiki