OpenPHDGuiding/phd2 #1216: Is SSAG the same that SSAGPro for deprecation?

koalazak, 2024-06-25:
Hello,

checking the changelog I saw that in `2.6.13dev1` version you removed the support for "SSAG" Cameras:

[image]

But in the Camera selection dropdown I am using a "SSAGPro" Camera:

[image]

Are they the same? I am out? :/

thank you

agalasso, 2024-06-26:
@koalazak no worries, the camera will continue to work in PHD2 via the ASCOM driver!

Only the native (non-ASCOM) camera driver for the [old SSAG camera](https://www.telescope.com/Orion-StarShoot-AutoGuider/p/52064.uts) has been removed since the driver library provided by Orion is years out of date and not compatible with our current development environment.

The SSAG Pro is a completely different camera despite the similar names. Generally speaking, any camera that has an ASCOM or INDI driver will work with PHD2.

koalazak, 2024-06-26:
perfect. thank you for the clarification!