OpenPHDGuiding/phd2 #1229: locals_update.sh seems to be broken in the recent master.

Eyeke2, 2024-07-29:
Maybe I miss something, but the script stopped working recently, possibly after relocation all source files to 'src' folder.

Eyeke2, 2024-07-29:
@agalasso Suggested fix

diff --git a/cmake_modules/PHD2BuildDoc.cmake b/cmake_modules/PHD2BuildDoc.cmake
index 2af3c603..83424d14 100644
--- a/cmake_modules/PHD2BuildDoc.cmake
+++ b/cmake_modules/PHD2BuildDoc.cmake
@@ -330,7 +330,7 @@ function(generate_single_doc_targets)
               -E make_directory
               "${current_translation_output_folder}"
           COMMAND
-            ${XGETTEXT} *.cpp *.h -C
+            ${XGETTEXT} src/*.cpp src/*.h -C
               --from-code=CP1252
               --keyword="_"
               --keyword="wxPLURAL:1,2"

agalasso, 2024-07-31:
@Eyeke2 thanks for tracking that down!

commit c60f5ea3 (2024-07-31): run xgettext in the src directory

update the xgettext command to find the source files since we moved them to src/

Fixes #1229
commit 6b9ca856 (2024-07-31): run xgettext in the src directory

update the xgettext command to find the source files since we moved them to src/

Fixes #1229
PR #1231 (2024-07-31): run xgettext in the src directory

# Based on PR #1230

## Full chain of PRs as of 2024-07-30

* PR #1231:
  `andy/build-locales-src-directory` ➔ `andy/remove-unused-file`
* PR #1230:
  `andy/remove-unused-file` ➔ `master`

run xgettext in the src directory

update the xgettext command to find the source files since we moved them to src/

Fixes #1229
commit c1c578a8 (2024-07-31): run xgettext in the src directory (#1231)

update the xgettext command to find the source files since we moved them to src/

Fixes #1229
closed 2024-07-31 by agalasso