OpenPHDGuiding/phd2 #1244: Dark Library Creation Taking Over Two Hours with QHY-L2C

19dexter86, 2024-10-05:
Hi everyone,

I'm having trouble with the dark frame library creation in PHD2. The process is taking over two hours, which seems unusually long. I have set the dark library to be created with exposure times between 1 and 6 seconds, with 5 frames each. My camera is a QHY-L2C, and I don't experience any issues with it otherwise. In other software, like SharpCap, the camera works perfectly fine, delivering a clean image, and I can see a high frame rate during capture.

In PHD2, I also don't encounter any issues when I start the loop – the images appear normally. It's just the dark frame library creation that takes an unusually long time.

Has anyone else experienced this issue, or does anyone have suggestions on how to resolve it?

Thanks in advance for your help!

19dexter86, 2024-10-05:
[PHD2_DebugLog_2024-10-05_091430.txt](https://github.com/user-attachments/files/17266609/PHD2_DebugLog_2024-10-05_091430.txt)

bwdev01, 2024-10-05:
This isn't a defect or shortcoming in PHD2, it's something specific to your setup.  Please post your problem on our support forum:
[https://groups.google.com/g/open-phd-guiding](url).  Also, please upload your log files to our server following the detailed instructions shown here:  [https://openphdguiding.org/getting-help/](url).   If you want to get a head-start on resolving the problem, update to the 2.6.13dev5 release of PHD2 and update your cameras drivers.  I'm going to close this development issue and look for your post on the support forum.  Regards.

closed 2024-10-05 by bwdev01