OpenPHDGuiding/phd2 #1245: Want bit mode selection for QHY cameras

daleghent, 2024-10-10:
The QHY SDK outputs image data in 8, 16, or 32 bit modes. The modes that are available depend on the camera model; the 32bit output mode can be ignored as that pertains to a few of their Eris platform cameras (QHY268/600/461/411) that must be specially configured to do so and under binning, which PHD2 would never need to do. 8 and 16 are what's available in all cases.

Currently, the QHY native driver in PHD2 outputs in 8bit mode as `SetQHYCCDBitsMode()` is never called during camera initialization to set up 16bit output. 

The proposal is to add a setup dialog to the QHY driver that offers radio button options for 8 or 16 bit output, with the default being 8 so as to maintain performance and dark library compatibility for existing users of this driver. Users may then switch to 16bit mode and remake their dark library on an elective basis.

daleghent, 2024-10-16:
WIP branch here https://github.com/daleghent/phd2/tree/qhy-improvements

daleghent, 2024-10-17:
WIP camera options window

[image]

bwdev01, 2024-10-17:
Hi Dale, this looks good at this point.  My question is whether we understand the peculiar behavior reported regarding the bits-per-pixel property.  In particular, I want to know why we’re constraining the MaxADU value to be 255.  That property is represented by a text field in the UI so the constraining mechanism should be coming from this code when the form is closed:

 

saturationVal = wxMin(val, SaturationValFromBPP(m_pCamera));

 

This suggests that a read of the bits-per-pixel value is returning 8 even though the camera is returning 16-bit data.  I didn’t see anything in your code that did anything in this area although perhaps I missed it.  AFAICT, the m_bpp variable in cam_qhy.cpp is being set correctly after the camera is connected so I don’t understand what’s wrong – and of course, this has worked fine for all the previous QHY cameras.  It may be worth remembering that we’re apparently dealing with a beta version of the camera driver here.

 

image.png (view on web) <https://github.com/user-attachments/assets/389934d9-1bab-446e-9a64-5698f6912661>

daleghent, 2024-10-18:
Reviewing [their logs](https://openphdguiding.org/logs/dl/PHD2_logs_3Lis.zip), the user was connecting and reconnecting with multiple cameras from QHY and ZWO, with both their respective ASCOM and native drivers. In short, the log contents were all over the place, as there were multiple profiles being switched to. This is what I've been able to distill:

1. Connected to QHY5III462M via QHY ASCOM driver; previous camera was "ZWO ASI290MM Mini" via native ZWO driver
  - BitsPerPixel=8
  - FullSize=(0,0)
2. Reconnected to the same, same image dimensions/bpp reported
3. Connected to QHY native driver which reported bpp=16. User kept `SaturationADU` at 255.

So, the QHY ASCOM driver, via the PHD2 ASCOM client, kept asserting a BPP of 8. Curious of where this came from, I went into the PHD2 ASCOM client code and found that it derives the `SaturationADU` parameter using the driver's `MaxADU` property. Since I have a QHY5III462C on my desk, I fired up the ASCOM Diagnostics app and used that to connect to a it via the QHY ASCOM driver and PHD2 reported the same BPP=8 result. Annoyingly, ASCOM Diagnostics doesn't dump all camera driver properties so I could not see what the driver was reporting for `MaxADU`, but it does dump the contents of the driver's ASCOM profile storage. To my surprise, there was a MaxADU variable associated stored in the profile with a value of `255`:

```
QHY5III462C-5764394f9f37839b2ElectronsPerADU           1
QHY5III462C-5764394f9f37839b2FullWellCapacity          65535
QHY5III462C-5764394f9f37839b2Gain                      100
QHY5III462C-5764394f9f37839b2HasCooler                 False
QHY5III462C-5764394f9f37839b2HasShutter                False
QHY5III462C-5764394f9f37839b2MaxADU                    255
QHY5III462C-5764394f9f37839b2MaxBinX                   2
QHY5III462C-5764394f9f37839b2MaxBinY                   2
```
I'm guessing that the QHY ASCOM driver uses the ASCOM Profile system as a per-camera key:value store, and all these camera parameters aren't dynamically derived but are actually presets! So I fired up the ASCOM Profile Explorer and found it:

[image]

So I edited the `QHY5III462C-5764394f9f37839b2MaxADU` field to have a value of `65535`. Reconnecting to the QHY ASCOM driver in PHD2, it now reports BPP=16:

```
21:32:14.070 00.001 210664 Connected Camera: QHY5III462C-5764394f9f37839b2 (ASCOM)
21:32:14.071 00.001 210664 FullSize=(0,0)
21:32:14.071 00.000 210664 PixelSize=2.90
21:32:14.072 00.001 210664 BitsPerPixel=16
21:32:14.072 00.000 210664 HasGainControl=0
21:32:14.072 00.000 210664 HasShutter=0
21:32:14.073 00.001 210664 HasSubFrames=1
21:32:14.073 00.000 210664 ST4HasGuideOutput=1
```

So it seems that the QHY ASCOM driver is hard-coded to report a `MaxADU` of `255` and only by editing the camera's entry in the driver's ASCOM profile storage area can you induce it to report a `MaxADU` of `65535`. Ha. So that explains the source of the 8bit saturation level.

The user's other complaint was that the native driver worked but it operating in 8bit mode. The logs clearly show it was in operating in 16bit mode; I think the user was confused that the saturation ADU was staying at 255 and not being automatically set to 65535. They were also switching PHD2 hardware profiles so there is that to consider - fog of confusion is more likely.

So this boils down to 2 things:
1. A faulty QHY ASCOM driver that always reports a `MaxADU` of `255` unless manually edited in the driver's ASCOM profile. This is a case of garbage-in, garbage-out from PHD2's perspective.
2. The QHY native driver doesn't make it clear which bit output the camera is operating in. The user was probably used to seeing this based on his experience with his ZWO camera, where the ZWO native driver implements a setup dialog that permits the user to directly specify 8 or 16bit output. 

Digging around in the QHY native driver, I found it would be useful to bring the driver code up to date with camera features, give the user some configuration options, and certainly directly assert the bit mode that the camera is expected to operate in.

agalasso, 2024-10-18:
thanks for that info @daleghent very interesting!
I agree it would be great to bring the driver code up to date, and we'd be happy to take your pr whenever you feel it is ready

daleghent, 2024-10-18:
@agalasso thanks. I think this means we should really warn people away from using the QHY ASCOM driver. As for the commits, I'm going to let this soak a little as I have some other time-critical stuff to finish over the next few days. When I post up the PR, do you want all the commits squashed?

agalasso, 2024-10-18:
> When I post up the PR, do you want all the commits squashed?

Either way is fine. When we merge we will squash automatically, and when reviewing I don't look at the individual commits anyway.   For big stuff that touches multiple areas that should be delivered incrementally, separate stacked prs are ideal (and we would request that). But I don't think that's the case here.

daleghent, 2024-10-25:
@agalasso  or @bwdev01 , Is there a handy way to get the index value of the selected camera before it's connected to? I want to refine the camera's setup options and enable/disable them in the UI in accordance with the selected camera's features, but it's not clear how I can access this information from with the driver's setup dialog in absence of a connection being active..

agalasso, 2024-10-26:
Unfortunately the index value is not a stable identifier: it can change if the camera is plugged in to a different USB port, and it I'm not even sure it is guaranteed to stay the same across machine reboots or plugging in other USB devices.
When the user connect to a camera we get the driver-provided camera id (typically a camera serial number -- see EnumCameras) and associate that with the Equipment Profile so that the same camera is connected to every time even if the index changes.
The upshot is that you should be able to just save whatever settings you need in the profile (`pConfig->Profle`). For an example `CameraToupTek::ShowPropertyDialog()`.
I don't think we have any precedent in any of the other cameras for enabling/disabling features in the ui before the camera is connected. One idea that comes to mind would be to connect (`OpenQHYCCD` -- not Camera::Connect) right when the settings dialog is opened and get access to whatever camera properties you need there.

daleghent, 2024-10-28:
Ah ok. QHY's SDK offers the ability to see if a camera of a given index number (as returned by its scan function) has a particular option available to it without having to open & initialize the camera first - `GetQHYCCDBeforeOpenParam()`. The tooltips for the options are worded to explain that a given option _may_ not be available on the camera in question. Perhaps this is something to revisit at a later point.

daleghent, 2024-11-22:
All work completed

closed 2024-11-22 by daleghent