OpenPHDGuiding/phd2 #1271: Request to correct the logic for setting m_pLaunchDarks control in profile_wizard.cpp

Eyeke2, 2024-12-28:
In the STATE_WRAPUP case block of profile_wizard.cpp, the line:
`m_pLaunchDarks->SetValue(m_useCamera || m_pLaunchDarks);`

appears problematic, as m_pLaunchDarks is both the target and part of the condition. This could lead to unintended behavior.
If the intended logic is to enable the control when either m_useCamera or another variable (e.g., m_launchDarks) is true:
`m_pLaunchDarks->SetValue(m_useCamera || m_launchDarks);`

To enable the control only when both are true:
`m_pLaunchDarks->SetValue(m_useCamera && m_launchDarks);`

Relevant section from the STATE_WRAPUP block:
```
case STATE_WRAPUP:
    SetTitle(TitlePrefix + _("Finish Creating Your New Profile"));
    m_pGearGrid->Show(false);
    m_pWrapUp->Show(true);
    m_pNextBtn->SetLabel(_("Finish"));
    m_pNextBtn->SetToolTip(_("Finish creating the equipment profile"));
    m_pLaunchDarks->SetValue(m_useCamera || m_pLaunchDarks);
    m_pInstructions->SetLabel(_("Enter a name for your profile and optionally launch the process to build a dark library"));
    m_pAutoRestore->Show(m_PositionAware || m_SelectedAuxMount != _("None"));
    m_pAutoRestore->SetValue(m_autoRestore);
    SetSizerAndFit(m_pvSizer);
    break;
```

Eyeke2, 2024-12-28:
@bwdev01 @agalasso

bwdev01, 2024-12-28:
Yes, it’s essentially a typo, it has the net effect of always initializing the checkbox to true.   I’ll fix it as part of some other changes that are pending.

 

Thanks,

Bruce

closed 2025-04-20 by bwdev01