OpenPHDGuiding/phd2 #1290: Problem with fetching OGMAcamSDK from a repo

Eyeke2, 2025-01-16:
During run_cmake process - when the script attempts to fetch the OGMAcamSDK repository, the git fetch operation fails when using commit **_5e83d81384479b1684b97e9f6936e453050da4cb_**. As a result, the run_cmake script cannot complete successfully. However, switching to the head commit **_9787aaf4882497262a8ba1612b43f93fb083d2e6_** resolves the issue, and the git fetch operation works as expected.

The error seen during build:

```
MSBuild version 17.12.12+1cce77968 for .NET Framework

  1>Checking Build System
  1>Creating directories for 'ogmacamsdk-populate'
  Performing download step (git clone) for 'ogmacamsdk-populate'
  Cloning into 'ogmacamsdk-src'...
  fatal: unable to read tree (5e83d81384479b1684b97e9f6936e453050da4cb)
  CMake Error at ogmacamsdk-subbuild/ogmacamsdk-populate-prefix/tmp/ogmacamsdk-populate-gitclone.cmake:49 (message):
    Failed to checkout tag: '5e83d81384479b1684b97e9f6936e453050da4cb'

```

satmonkey, 2025-01-21:
same issue here

agalasso, 2025-01-22:
@Eyeke2 please disregard my precious comment.  I am now able to reproduce the problem and will commit a fix shortly.

commit a6a55fcd (2025-01-22): Fix build error due to invalid CMake function parameter

`GIT_SHALLOW` is incomaptible with `GIT_TAG` being a SHA
for FetchContent_Declare()

Fixes #1290
PR #1291 (2025-01-24): Fix build error due to invalid CMake function parameter
`GIT_SHALLOW` is incomaptible with `GIT_TAG` being a SHA for FetchContent_Declare()

Fixes #1290
commit 7454c9a4 (2025-01-24): Fix build error due to invalid CMake function parameter (#1291)

`GIT_SHALLOW` is incomaptible with `GIT_TAG` being a SHA
for FetchContent_Declare()

Fixes #1290
closed 2025-01-24 by agalasso